espruino/EspruinoDocs #114: I2C.setup() and SPI.setup() documentation in espruino.com/Reference don't list default baud/bitrate

SpenceKonde, 2015-01-28:
Sure, I could ask, and I remember Gordon saying the default for I2C was 50khz (50000), but AFAIK, nobody's ever asked this about SPI, so god only knows what the default speed for that is. 

This is the kind of information that should be listed in the reference, along with the minimum and maximum possible. It's not clear to me how these reference pages are generated, and whether I should be reporting this issue here, or in the main espruino repo.

gfwilliams, 2015-01-29:
The reference pages are auto-generated from comments in the code - eg: https://github.com/espruino/Espruino/blob/master/src/jswrap_spi_i2c.c#L144

Looks like default for SPI is 100,000. I'll update the docs. minimum and maximum are iffy as they depend on the device and clock speed (which I may make tweakable on the Pico). I suppose the more sensible think might be to warn if it can't get within +/- 50%

SpenceKonde, 2015-01-29:
How about setting the SPI's _options.baud to the actual baud that was set, and so if we care, we can check it? (eg SPI2._options.baud)

gfwilliams, 2015-02-02:
Yes, that could be a good plan I guess.

gfwilliams, 2015-07-08:
Just to add, this is now in the docs - but the baud rate isn't to up in the code as suggested

gfwilliams, 2016-03-10:
This is actually in `SerialX._baudrate` now

closed 2016-03-10 by gfwilliams