espruino/EspruinoDocs #120: Issue with MFRC552 Module

MrTimcakes, 2015-03-08:
When uploading the code to the Espruino i receive the error
```
Uncaught Error: Field or method does not already exist, and can't create it on undefined
 at line 1 col 190
...("MRFC522 Request Error "+a);a=this.r(20);return this.ra(18,...
                               ^
in function "req" called from line 1 col 35
{this.w(26,7);return 0<this.req(38).length}
                                   ^
 at line 1 col 36
{this.w(26,7);return 0<this.req(38).length}
                                    ^
in function "isNewCard" called from line 1 col 17
{this.isNewCard()&&a(this.getCardSerial())}
                 ^
```

gfwilliams, 2015-03-09:
Thanks - this looks like it's actually a problem with exceptions.

I *think* that your module is probably wired up wrong. It's actually not getting a response so is trying to throw the error `MRFC522 Request Error`. Unfortunately due to an Espruino bug it seems to be getting overwritten with `Field or method does not already exist`.

gfwilliams, 2015-03-26:
Just to add that Espruino 1v76 and later will have a fix in, so the error should display properly.

Did you manage to fix this by rewiring I2C? - maybe by adding pullups?

closed 2015-04-15 by gfwilliams