espruino/EspruinoDocs #127: ESP8266 getIP() returns incorrect result

SpenceKonde, 2015-04-14:
I'm getting 192.168.4.1 as the IP when I do wifi.getIP(). 

The IP address that the ESP8266 actually gets is 192.168.2.102. It seems that it's getting the IP that it would use when acting as an access point, rather than the IP it has as a client.

gfwilliams, 2015-04-14:
Hmm. Are you using the same esp8266 firmware? I remember looking at this
and was pretty sure it worked. The at command returns 2 IPs when the ap
mode is 3, but was pretty sure it just returned a single one if the mode
was 1 or 2 (which is all the driver should set it to).

- Gordon

SpenceKonde, 2015-04-14:
This was with f/w version 0018000902-AI03, using the example code...

gfwilliams, 2015-04-14:
Thanks - found a bit of a copy/paste bug :)

commit 6dad3fe2 (2015-04-14): fix getIP changing CWMODE, when connectAP should do it - fix #127
closed 2015-04-14 by gfwilliams