espruino/EspruinoDocs #130: tutorials/Pico CPU Monitor.md

AxelKlarmann, 2015-04-17:
Hey there,
just checked the nice gadget on Mac OSX 10.9

The command "paste" failed as it was in need of the filename. If you give "-" as filename it will read the piped data.

As i'm german it was difficult to find another problem, because the decimal point in germany is "," not "." and bc was not able to calculate … maybe a little hint to mention for non experts.

There was a little typo, because the device was missing a "b" in "cu.usmodem"

The complete command that worked for me, besides setting my locale to english, was:
echo -e "draw(`ps -A -o pcpu | tail -n+2 | paste -sd+ - | bc`)\n" > /dev/cu.usbmodem*;

Thanks again and awesome work!

gfwilliams, 2015-04-20:
Great, thanks for the report! I'll make sure I update the documentation. I should be able to make the code convert `,` to `.` on the Pico itself

gfwilliams, 2015-05-18:
Actually fixed this a while back...

closed 2015-05-18 by gfwilliams