espruino/EspruinoDocs #141: tasks/File Converter.md allows files to be larger then 20KiB

joppiesaus, 2015-05-16:
Today I wanted to convert a large file, but of course that didn't work because it was simply to large.
The console had thrown me a `Uncaught RangeError: Maximum call stack size exceeded`. The file was larger then 20KiB and it didn't leave the "File too long - must be less than 20kB".
That's because it still executes `var b64Str = 'atob("'+btoa(String.fromCharCode.apply(null, bytes))+'")';`, when this is too large js will just crash.
I'll make a pull request that fixes this.

gfwilliams, 2015-05-18:
Argh, Thanks!

Actually I should try and improve that code anyway - just made a bug for it: https://github.com/espruino/EspruinoDocs/issues/144

Thanks for the fix!

gfwilliams, 2015-05-18:
Actually, could you post up your code? I'm surprised you got the call stack error... Usually it'd just be a normal 'out of memory'...

closed 2015-05-18 by gfwilliams

joppiesaus, 2015-05-18:
@gfwilliams :+1: 
I'm not sure what you mean, but I got the Uncaught RangeError at the `var b64Str = 'atob("'+btoa(String.fromCharCode.apply(null, bytes))+'")';`
(https://raw.githubusercontent.com/espruino/EspruinoDocs/3ca286496c2ea12214223b0b5cc6efaa91c96df9/tasks/File%20Converter.md)[in FileConverter.md line 59

gfwilliams, 2015-05-18:
Oh, ok - it was actually an error on your PC rather than in Espruino? :)

joppiesaus, 2015-05-18:
Affirmative

commit 6e529f3d (2015-05-18): Fixes #141

:snake:
commit ddafa50c (2015-05-18): Merge pull request #142 from joppiesaus/patch-1

Fixes #141