espruino/EspruinoDocs #145: OSC support

gfwilliams, 2015-06-03:
_From @Jorgen-VikingGod on June 3, 2015 12:22_

would be nice to have OSC support library on Espruino
like this:
https://github.com/CNMAT/OSC

more about this protocoll:
http://opensoundcontrol.org/

_Copied from original issue: espruino/Espruino#542_

gfwilliams, 2015-06-03:
_From @loop23 on June 3, 2015 12:39_

+1
I may give it a try, I'm into sound protocols and would like to study OSC for personal sake

gfwilliams, 2015-06-03:
Just moved to here, because this can be done as a module rather than as part of the Espruino firmware...

Good idea though.

loop23, 2015-06-03:
Yes, I was going to do it as a module anyway.
The event emitter interface is going to be very useful I believe

loop23, 2015-06-03:
Probably could just port https://www.npmjs.com/package/osc ; Are there any guidelines/tips for "porting node stuff to espruino"?

gfwilliams, 2015-06-03:
Looks good - it doesn't seem like it's that bloated, it may run with quite minimal changes.

I tips really - just try and get rid of external dependencies (of which there aren't many) and try and avoid having loads of 'stub' functions like `function getX() { return X; }`

Jorgen-VikingGod, 2015-06-04:
you guys are awesome. 
btw. is there also a DMX module available - or still missing?
thx @loop23 to try/port this. Perhaps I can help on testing

gfwilliams, 2015-06-05:
There's no DMX library at the moment I'm afraid. There are some posts about it on the forum though - I think it'd be possible to get a transmitter going very easily with just a few lines of code. Receive could be a bit more tricky as you have to detect the start of frame - someone would have to try it and see.

Jorgen-VikingGod, 2015-06-05:
I currently working on a simple ArtNet Node based on ESP8266 + MAX485 chip. We have some lights at our hackerspace controlable via DMX, but still not enough cables (fucking expensive these cables).
So my idea is to build simple WiFi-ArtNet-Nodes (DMX over LAN = ArtNet)
I think it is really simple to build - you just need a couple of components; the ESP8266 WiFi SoC + MAX485 TTL chip, e.t.c.
Put everything into 3D printed case with antenna and connect DC power jack. Plugin into DMX device and wooohoo: control over air :D 

There are a lot of arduino libraries out there doing this stuff. 
Perhaps I will try to port this also to the Espruino.

loop23, 2015-06-10:
I did some preliminary work porting  https://www.npmjs.com/package/osc  and it's not looking too good; It runs out of memory when uploading. I removed all write functions, and at least it fits (but it's missing the DataView). I'll try and see if I find a polyfill of sorts.

EricEspinoza, 2017-08-31:
Hey @loop23, I know it's been a while, but did you manage to get osc working on a Espruino device? And if not, what problem did you run into that halted your progress? Any luck on your end, @gfwilliams ?

gfwilliams, 2017-09-01:
I haven't had time, however there is now DataView support in Espruino, which should make the porting process easier!

loop23, 2017-09-01:
No, sorry, i didn't have a chance to get back to it. Be sure to have more memory than standard espruino if you want to use both in and out at the same time.

closed 2024-02-09 by gfwilliams