espruino/EspruinoDocs #152: connecting error at ESP8266 version:0.25.0.0

bathtimefish, 2015-08-09:
Hi, I tried [sample code of ESP8266](http://www.espruino.com/ESP8266) on espruino pico (wied to Serial1 B6, B7)

```
Serial1.setup(115200, { rx: B7, tx : B6 });
var wifi = require("ESP8266WiFi").connect(Serial1, function(err) {  
  if (err) throw err;
  wifi.reset(function(err) {
    if (err) throw err;
    console.log("Connecting to WiFi");
    wifi.connect("BTFNW-g","9ea5f5c95d057", function(err) {
      if (err) throw err;
      console.log("Connected");
      // Now you can do something, like an HTTP request
      require("http").get("http://www.pur3.co.uk/hello.txt", function(res) {
        console.log("Response: ",res);
        res.on('data', function(d) {
          console.log("--->"+d);
        });
      });
    });
  });
});
```

I got error like this:

```
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v80 Copyright 2015 G.Williams
>echo(0);
=undefined
Connecting to WiFi
>Uncaught WiFi connect failed: WIFI DISCONNECT
 at line 2 col 22
      if (err) throw err;
                      ^
in function "b" called from line 1 col 37
{"OK"!=a?b("WiFi connect failed: "+a):b(null)}
                                     ^
in function "b" called from line 1 col 26
{c=void 0;var d;b&&(d=b(a))?(c=p,b=d):clearTimeout(e);void 0...
                          ^
in function "c" called from line 1 col 325
...f&&(e[f](k),n=!0);n||c&&c(k)}a=a.substr(d+1);"\n"==a[0]&&(a=...
                               ^
in function called from system
Uncaught WiFi connect failed: undefined
 at line 2 col 22
      if (err) throw err;
                      ^
in function "b" called from line 1 col 37
{"OK"!=a?b("WiFi connect failed: "+a):b(null)}
                                     ^
in function "b" called from line 1 col 16
{c=void 0;b&&b()}
                ^
in function called from system
```

So I comfirmed firmware version

```
function test(serial, baud, pins) {
  serial.removeAllListeners();
  var l="";
  serial.on('data', function(d) { l+=d;});
  serial.setup(baud, pins);
  serial.write("AT+GMR\r\n");
  setTimeout(function(){console.log(JSON.stringify(l));},100);
}
test(Serial1, 115200, { rx: B7, tx : B6 });
```

results:

```
>echo(0);
=undefined
"AT+GMR\r\r\nAT version:0.25.0.0(Jun  5 2015 16:27:16)\r\nSDK version:1.1.1\r\ncompile time:Jun  5 2015 21:03:10\r\n\r\nOK\r\n"
> 
```
I tried a test code like this.

```
function testESP8266(serial, baud, pins) {
  serial.removeAllListeners();
  serial.on('data', function(d) { console.log('onData: ' + JSON.stringify(d)); });
  serial.setup(baud, pins);
  var at = require('AT').connect(serial);
  //at.debug();
  at.cmd("ATE0\r\n", 1000, function(d) {
    if(d=='OK') {
      at.cmd("AT+CWMODE=1\r\n", 1000, function(d) {
        console.log(d);
        if(d=='OK') {
          at.cmd("AT+CWJAP=" + "\"[SSID]\"" + "," + "\"[PASS PHRASE]\"\r\n", 20000, function(d) {
        console.log("!!! CWJAP result: " + d);
      });
        }
      });
    }else {
      console.log("error at ATE0");
    }
  });
}
testESP8266(Serial1, 115200, { rx: B7, tx : B6 });
```

results:

```
>echo(0);
=undefined
onData: "\r\nOK"
onData: "\r\n"
onData: "\r\nOK\r\n"
OK
onData: "W"
onData: "IFI CONNECTED\r\n"
!!! CWJAP result: WIFI CONNECTED
onData: "W"
onData: "IFI GOT IP\r\n"
onData: "\r"
onData: "\nOK\r\n"
> 
```

After ran this code, I could get IP address in CLI of Espurino IDE.

```
>Serial1.write("AT+CIFSR\r\n");
=undefined
onData: "+CIFSR:STAIP,\"192.168.1.21\"\r\n"
onData: "+CIFSR:STAMAC,\"18:fe:33:a5:23:ba\"\r\n\r\nO"
onData: "K\r\n"
```

at.cmd("AT+CWJAP=...") returns "WIFI DISCONNECT", I think that it is the cause of the error .

What's wrong with my thinking?

gfwilliams, 2015-08-10:
Hi, just wondering, did you post in the forum too? Or was that someone with a similar issue?

Please can you try:

```
var wifi = require("ESP8266WiFi_0v25").connect(Serial1, function(err) {  
```

instead of:

```
var wifi = require("ESP8266WiFi").connect(Serial1, function(err) {  
```

I think it's because you have the newer firmware, and so need to use a slightly different module in order to connect to the ESP8266. I'll try and update the documentation...

gfwilliams, 2015-08-10:
... also, thanks for all your support - both the KickStarters, and all the great stuff you've published about Espruino in the last few years!

bathtimefish, 2015-08-10:
Thank you @gfwilliams 

> var wifi = require("ESP8266WiFi_0v25").connect(Serial1, function(err) {  

Yes! I got works fine from that!
I did not find the forum sorry...

Actually, This post might was good for me..
http://forum.espruino.com/conversations/270274/

Thank you for your advice & Thank you for awesome device Espruino!
I have wanted to know espruino to my country developers :)

gfwilliams, 2015-08-19:
Closing this now...

closed 2015-08-19 by gfwilliams