espruino/EspruinoDocs #158: tutorials/Quick Start.md - missing semicolons in onInit() example

programmicha, 2015-09-12:
function onInit() {
  digitalWrite([LED1,LED2], 2);
  setTimeout("digitalWrite([LED1,LED2], 1);", 1000)
  setTimeout("digitalWrite([LED1,LED2], 0);", 2000)
}

has missing semicolons (the web ide shows this as well)

should be

function onInit() {
  digitalWrite([LED1,LED2], 2);
  setTimeout("digitalWrite([LED1,LED2], 1);", 1000);
  setTimeout("digitalWrite([LED1,LED2], 0);", 2000);
}

programmicha, 2015-09-12:
when you fix this, please add some doc to the code as well

```digitalWrite([LED1,LED2], 2); // 2 is interpreted as an binary array (read: [11]), so each LED gets a 1 for "activate"```

```setTimeout("digitalWrite([LED1,LED2], 1);", 1000);``` is a bit confusing, after reading the reference for setTimeout i expected 
```setTimeout(digitalWrite, 1000, [LED1,LED2], 1); ```(which works) i know it's possible in javascript to use both styles here, but for beginners like me it is a bit confusing

gfwilliams, 2015-09-14:
Thanks! Yes, I'll add some documentation about that too.

gfwilliams, 2015-09-14:
Ok, just done - it'll update next time I push to the website, this week sometime

commit 43a7a3d7 (2015-09-14): Fix dodgy Quick Start code, and add some more explanation for it (fix #158)
closed 2015-09-14 by gfwilliams