espruino/EspruinoDocs #163: more/detailed description for current the module loading mechanism

programmicha, 2015-10-05:
coming from http://forum.espruino.com/conversations/275060/#12543587

# what we have

* http://www.espruino.com/Modules 
  * seems to be the place to look for informations about concrete modules and abstract handling of module loading 
  * describes the "standard" module loading mechanism, (web ide loads minified module from internet)
  * mentions sd-card loading mechanism on 2 places ("stand-alone" and FAQ )
  * announcement of coming feature - pico loads module by itself, if it is wifi enabled

# what's missing

* proper information structure
  * there should be a central part for the concrete modules (already there)
  * and another central part for "working with modules"
* more informations on:
  * exactly where are the modules loaded from?
  * minifying, why, when, when not
  * to my surprise it is not the modules folder, but devices
    * but some "modules" are loaded from the modules folder too e.g. the AT module
    * some modules are not listed on the overview, e.g. the AT module
  * load specific module "versions" (aka github commits)
    * see http://forum.espruino.com/comments/12543504/
  * load modules from web side project modules folder
    * priority? internet vs. local folder
  * loading custom modules

# what i propose

* a pragmatic approach
  * divide existing http://www.espruino.com/Modules into two logical parts, but keep all at one page
    * module listing
    * **working with modules** with the needed informations 
  * on each module page a backlink to the module overview - check, afaik all pages already have this

gfwilliams, 2015-10-05:
Sounds like a good plan, it definitely needs filling in.

Actual `Modules.md` is here: https://github.com/espruino/EspruinoDocs/blob/master/Modules.md

> exactly where are the modules loaded from?

espruino.com/modules - *but the url can be configured in Web IDE settings*

> minifying, why, when, when not

To save space. By default, minified file is loaded if it exists, then the unminified.
And unminified files are then minified by the Web IDE (it's another setting)

> to my surprise it is not the modules folder, but devices

Modules for hardware ('devices') are in `devices`, Modules that are basically pure software are in `Modules`. 

> but some "modules" are loaded from the modules folder too e.g. the AT module

All modules (from wherever they are in EspruinoDocs) go into espruino.com/modules

> some modules are not listed on the overview, e.g. the AT module

I think it's because it doesn't have a `.md` file. We should probably add one.

> load specific module "versions" (aka github commits) see http://forum.espruino.com/comments/12543504/

Yes, definitely needs some filling in.

> load modules from web side project modules folder priority? internet vs. local folder

Local has priority

> loading custom modules

Yeah - we should cover developing using GitHub as the URL. It's probably the easiest to get set up and you have version history.

It makes sense to also add to/reference http://www.espruino.com/Writing+Modules as well

With the backlink, rather than being explicit I'd suggest I just add a bit of JS to espruino.com that auto-backlinks H1 elements.

gfwilliams, 2015-10-05:
... and to add, I also need to auto-anchor headings. It's amazingly frustrating not being able to `#link` to them

programmicha, 2015-10-05:
for the "load from github" feature:

after looking into https://github.com/espruino/EspruinoTools/blob/gh-pages/plugins/getGitHub.js it looks like you can just load any file from github which contains javascript, filename extension e.g. .js does not matter

gfwilliams, 2015-10-06:
Yes, absolutely...

programmicha, 2015-10-06:
thx, next info 

"for loading from local, beware of a 2 dimensional priority which combines online/offline and minified/ not minified" (can be changed in webide)

require(...) standard loading order
* local minified
* online minified
* local normal
* online normal

resource: http://forum.espruino.com/comments/12551225/

programmicha, 2015-10-10:
you can sneak peak on the current state on my fork under https://github.com/programmicha/EspruinoDocs/blob/EspruinoDocs-issue-163/Modules.md 

one problem i hit a lot here is some documentation on the Web IDE as well, but lets make one step after another

gfwilliams, 2015-10-12:
Hey, thanks! It looks great!

Might be worth a note that when enabling NPM modules, it:

* Only works if the module contains a single file
* Can cause some confusion with Espruino's modules, for instance `clock`.

Please can you keep the H1 and H2 titles in the original style (`===` and `---`) though? That's how I do it for the rest of EspruinoDocs, so it makes sense to use it for this too -  the automated tools I use also rely on the format of at least the first title so they can work out the page's title correctly.

programmicha, 2015-10-12:
thanks for the feedback Gordon, i added the NPM details and changed the markdown header style for H1 and H2 back to the original 
oh and feel free to correct the english, it is not my native language but i am willing to learn :-)

gfwilliams, 2015-10-13:
It seems great to me!

The formatting looks a bit odd (according to GitHub anyway) - I think it doesn't like it when you put 3 backticks, code, 3 backticks, all on a brand new line?

programmicha, 2015-10-13:
darn i tested the markdown preview local (Mou app) only, thx for the hint, i changed and tested it

gfwilliams, 2015-10-14:
Looks great - do you want to do a PR then?

PR #169 (2015-10-14): Espruino docs issue 163
fixes #163
closed 2015-10-14 by gfwilliams