espruino/EspruinoDocs #188: Build Process Not Working

chalkers, 2015-11-11:
With this version of node.js - v4.1.1

With these page dependancies:
```
    "acorn": "^2.6.2",
    "express": "^4.13.3",
    "highlight.js": "^8.9.1",
    "marked": "^0.3.5",
    "tern": "^0.16.0"
```

Running the build I get the following output:

```
$ node bin/build.js
Example File examples/distance_sensing_robot.js
Example File examples/http_file_server.js
Example File examples/http_get_image.js
Example File examples/led_vu_meter.js
Example File examples/rgb123_reaction_timer.js
Example File examples/simple_data_logger.js
Example File examples/slideshow_ili9341.js
Example File examples/temperature_display_pcd8544.js
Example File examples/temperature_graph_pcd8544.js
Example File examples/useless_box.js
Example File examples/wifi_humidity.js
Example File examples/wii_remote_control_helicopter.js
Example File examples/ws2811_led_clock.js
REJECTED devices/ADNS5050.md from Tutorials.md because of '-module' keyword
REJECTED devices/ADS1X15.md from Tutorials.md because of '-module' keyword
REJECTED devices/ADS7843.md from Tutorials.md because of '-module' keyword
REJECTED devices/ADXL335.md from Tutorials.md because of '-module' keyword
REJECTED devices/ADXL345.md from Tutorials.md because of '-module' keyword
REJECTED devices/RGBLed.md from Tutorials.md because of '-module' keyword
REJECTED devices/BH1750.md from Tutorials.md because of '-module' keyword
REJECTED devices/BMP085.md from Tutorials.md because of '-module' keyword
REJECTED devices/CAP1188.md from Tutorials.md because of '-module' keyword
REJECTED devices/Charlieplex.md from Tutorials.md because of '-module' keyword
REJECTED devices/DCF77.md from Tutorials.md because of '-module' keyword
REJECTED devices/DHT11.md from Tutorials.md because of '-module' keyword
REJECTED devices/DHT22.md from Tutorials.md because of '-module' keyword
REJECTED devices/DS18B20.md from Tutorials.md because of '-module' keyword
REJECTED devices/DS3231.md from Tutorials.md because of '-module' keyword
REJECTED devices/Digole.md from Tutorials.md because of '-module' keyword
REJECTED devices/EasyVR.md from Tutorials.md because of '-module' keyword
REJECTED devices/HC-SR04.md from Tutorials.md because of '-module' keyword
REJECTED devices/HD44780.md from Tutorials.md because of '-module' keyword
REJECTED devices/HMC5883.md from Tutorials.md because of '-module' keyword
REJECTED devices/HTU21D.md from Tutorials.md because of '-module' keyword
REJECTED devices/RFM69.md from Tutorials.md because of '-module' keyword
REJECTED devices/AT24.md from Tutorials.md because of '-module' keyword
REJECTED devices/ILI9163.md from Tutorials.md because of '-module' keyword
REJECTED devices/ILI9341.md from Tutorials.md because of '-module' keyword
REJECTED devices/KeyPad.md from Tutorials.md because of '-module' keyword
REJECTED devices/LPD6416.md from Tutorials.md because of '-module' keyword
REJECTED devices/MAX31855.md from Tutorials.md because of '-module' keyword
REJECTED devices/MAX7219.md from Tutorials.md because of '-module' keyword
REJECTED devices/MCP23xxx.md from Tutorials.md because of '-module' keyword
REJECTED devices/MCP4xxx.md from Tutorials.md because of '-module' keyword
REJECTED devices/MCP4xxx0.md from Tutorials.md because of '-module' keyword
REJECTED devices/MCP9808.md from Tutorials.md because of '-module' keyword
REJECTED devices/MFRC522.md from Tutorials.md because of '-module' keyword
REJECTED devices/MPL115A2.md from Tutorials.md because of '-module' keyword
REJECTED devices/MPL3115A2.md from Tutorials.md because of '-module' keyword
REJECTED devices/MPU6050.md from Tutorials.md because of '-module' keyword
REJECTED devices/MQTT.md from Tutorials.md because of '-module' keyword
REJECTED devices/Midi.md from Tutorials.md because of '-module' keyword
REJECTED devices/NRF24L01P.md from Tutorials.md because of '-module' keyword
REJECTED devices/Brain.md from Tutorials.md because of '-module' keyword
REJECTED devices/DS2xxx.md from Tutorials.md because of '-module' keyword
REJECTED devices/PCD8544.md from Tutorials.md because of '-module' keyword
REJECTED devices/PN532.md from Tutorials.md because of '-module' keyword
REJECTED devices/Encoder.md from Tutorials.md because of '-module' keyword
REJECTED devices/SH1106.md from Tutorials.md because of '-module' keyword
REJECTED devices/AT25.md from Tutorials.md because of '-module' keyword
REJECTED devices/SSD1306.md from Tutorials.md because of '-module' keyword
REJECTED devices/SSD1351.md from Tutorials.md because of '-module' keyword
REJECTED devices/STM32F1Flash.md from Tutorials.md because of '-module' keyword
REJECTED devices/SHT1x.md from Tutorials.md because of '-module' keyword
REJECTED devices/Servo Motors.md from Tutorials.md because of '-module' keyword
REJECTED devices/MemoryLCD.md from Tutorials.md because of '-module' keyword
REJECTED devices/SmartNixie.md from Tutorials.md because of '-module' keyword
REJECTED devices/TCS3200.md from Tutorials.md because of '-module' keyword
REJECTED devices/TCS3472x.md from Tutorials.md because of '-module' keyword
REJECTED devices/TSL2561.md from Tutorials.md because of '-module' keyword
REJECTED devices/Thermistor.md from Tutorials.md because of '-module' keyword
REJECTED devices/Wii.md from Tutorials.md because of '-module' keyword
REJECTED devices/XBee-API1.md from Tutorials.md because of '-module' keyword
APPEND_PINOUT /Users/andrew/Projects/EspruinoDocs/html/boards/EMW3165.html
fs.js:549
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                                                  ^

Error: ENOENT: no such file or directory, open '/Users/andrew/Projects/EspruinoDocs/html/boards/EMW3165.html'
    at Error (native)
    at Object.fs.openSync (fs.js:549:18)
    at Object.fs.readFileSync (fs.js:397:15)
    at /Users/andrew/Projects/EspruinoDocs/bin/build.js:437:22
    at Array.forEach (native)
    at Object.<anonymous> (/Users/andrew/Projects/EspruinoDocs/bin/build.js:343:15)
    at Module._compile (module.js:434:26)
    at Object.Module._extensions..js (module.js:452:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
```

gfwilliams, 2015-11-11:
Have you tried just running `build.sh`? That runs a few other things first, and also kicks off the module minification, which might be what you want?

IIRC you also need `Espruino` checked out at the same level as `EspruinoDocs`

chalkers, 2015-11-11:
Ahh I missed out the checking out of `Espruino` step.

Doing `./build.sh` results in this...
```
Compilation failed - function wrapper not found. Got: "Error(22): T" "ater.\n"
Error(22): Too many compiles performed recently.  Try again later.

Sending to Google Closure Compiler...
Minification complete!
Error(22): Too many compiles performed recently.  Try again later.

devices/MPL115A2.js compile FAILED../
```

However `node bin/build.js` works now. The express app works now too.

gfwilliams, 2015-11-12:
Ok, great! Just run `build.sh` again tomorrow and it should be fine hopefully :) I use the online closure compiler service because it's easier, but it is rate limited.

After you've done it initially you won't ever hit it again, because it will only compile the modules that have changed.

I can't actually remember where the minified modules end up - you might actually need a directory `espruinowebsite/www/modules` as well...

gfwilliams, 2016-03-10:
Closing this as seems to be sorted?

closed 2016-03-10 by gfwilliams