espruino/EspruinoDocs #195: Module RGBLed.js missing feature

yerpj, 2015-11-26:
I don't know if I can use this issue tracking system for discussions related to modules. If there is a better place to post such issues, please let me know.

This issue is about the module "RGBLed.js" 
see http://www.espruino.com/RGBLed for doc,
and https://github.com/espruino/EspruinoDocs/blob/master/devices/RGBLed.js for module itself

This module assumes that LEDs are driven HIGH, which is not common for many RGB leds components. I suggest adding a function that allows to invert the polarity of the drive.
This consists of adding a private parameter and a function:
```
function rgbled(pins, state, color) {
  ...
  this.inverted=false;
  ...
}
```
```
rgbled.prototype.invert = function (x) {
  this.inverted=x;
};
```

Then, the setColor function can be slightly modified this way: 
```
rgbled.prototype.setColor = function (color) {
  for (var i = 0, s = -6; i < 3; i += 1, s += 2) {
	if(!this.inverted){  
      this.rgbAnalog[i] = E.clip(parseInt(color.substr(s, 2), 16), 0, 255) / 255;
	}
    else {
      this.rgbAnalog[i] = 1-(E.clip(parseInt(color.substr(s, 2), 16), 0, 255) / 255);
	}
  }
  this._write();
};
```

It was successfully tested.

Another cool feature I also implemented is a breathe-like function (color changing slowly based on a sinusoidal function exectued on R,G and B components):
```
rgbled.prototype.breathe=function(){
	this._stop();
    this.state = true;
	var that=this;
	this.intervalId=setInterval(function(){
		var t=getTime();
		t=t%6.28318;
		var r=128+127*Math.sin(t);
		var g=128+127*Math.sin(t+2.09);
		var b=128+127*Math.sin(t+4.18);
		var rs=Math.floor(r).toString(16);
		if(rs.length<2)rs="0"+rs;
		var gs=Math.floor(g).toString(16);
		if(gs.length<2)gs="0"+gs;
		var bs=Math.floor(b).toString(16);
		if(bs.length<2)bs="0"+bs;
		var Color="#"+rs+gs+bs;
		that.setColor(Color);
	},100);
}
```

This new feature has also been tested and validated.

gfwilliams, 2015-12-02:
Thanks - Sorry about the delay.

I'll leave this bug open so I remember to tweak the documentation as well.

yerpj, 2015-12-02:
It is a good idea. I don't know how how many guys are using this module, but as I did not add the rgbled.prototype.breathe() in the pull request, leaving this bug open gives a chance for anyone to access the implementation of the latter.

commit 81dab843 (2020-11-05): Fix #195
closed 2020-11-05 by gfwilliams