espruino/EspruinoDocs #197: tutorials/Flashing Lights.md

jbchafer, 2015-12-05:
There are all kinds of mistakes. Has anybody tested your tutorial?

For example, Math.random() returns: -0.5 < value < 0.5

So your example:
setInterval(function() {
  digitalWrite(LED1, Math.random()>0.5);
}, 20);

Never turns on the LED.
Even the reference documentation is wrong about Math.random().

gfwilliams, 2015-12-05:
Are you actually using an up to date firmware? Math.random got broken for a few firmware versions, but it is definitely fixed in 1v83.

What other problems did you have?

gfwilliams, 2015-12-07:
Just tried all the bits of code on 1v83 and they seem to work fine. What were the 'all kinds of mistakes' you came across?

jbchafer, 2015-12-08:
I’m sorry. I apologize for my rash remark from a beginner on a new platform. It is working well now and very cool.

I updated the firmware (1v83, and 1v84) and Math.random works correctly.

The first time through the Quick Start, I notice the ‘Send to Web IDE’ button on the right side of all of the snippets of code. But I needed to append the next snippet and the button does a replace. My reflexes kept hitting the 'Send to Web IDE’ button, and therefore deleting the previous snippet. Must select, copy, switch app, select right panel, paste.

The Web IDE is nice. Console in one panel, editor in the other panel. I’ve used compilers where you through your code over the wall, and interpreters where you can iteratively build and test. This seems like a hybrid. But I’m looking for a ‘list’ command that is not there. If I can get the interpreter to display one function (edit (‘fname’), why not all/everything? The editor seems vulnerable to deleting work when those pesky 'Send to Web IDE’ buttons are pressed.

I appreciate your quick and patient response.

 — John Schafer

gfwilliams, 2015-12-09:
Thanks! You mean like this?

[image]

It's a good point - I should get rid of those when the code should just be put in the console. I'll add an issue for it.

To list everything in the interepreter, type `dump()`. It won't be perfect, as Espruino has to 'reconstruct' some things - when code gets sent it is executed right away, so for instance if you upload `a=1+2`, what actually gets stored is `a=3`.

commit 8ce757ce (2016-03-10): modify flashing lights tutorial to remove 'upload' button in cases where it doesn't make sense - fix #197
closed 2016-03-10 by gfwilliams