espruino/EspruinoDocs #205: Websockets code forces sent data to be JSON parseable

gfwilliams, 2015-12-16:
Here: https://github.com/espruino/EspruinoDocs/blob/master/modules/ws.js#L170

@samehhady is this required? It doesn't really seem to be at least on the web browser side... 

I notice we've got `broadcast` and `join` as well that don't really seem needed. Could we not have those wrapped up in another module like `wschat` or something?

samehhady, 2015-12-18:
@gfwilliams I wanted to provide the same functionality as the nodeJS ws module, if it is not usable i can bring back the older simple version

gfwilliams, 2015-12-18:
I'm not quite sure what the node.js `ws` module does. Here:

https://github.com/websockets/ws/blob/master/lib/WebSocket.js#L233

It looks a bit like if you send data in a 'raw' form, it might get sent as-is...

Also, seems like `broadcast` and `join` might just be examples, rather than being part of the actual websocket? https://github.com/websockets/ws#server-sending-broadcast-data

I just tried creating a websocket and didn't see any `join` or `broadcast` methods on it - did you add them because some other bit of code was using them?

samehhady, 2015-12-18:
Yup its more like an example or some sort of extend to the module which in our case wont be easy for other guys to do unless they modify the module itself, thats why i thought to add it. I believe you are right, I will remove this part and will keep it raw output and input.

gfwilliams, 2015-12-18:
Thanks! I think it won't be too hard for people to add to it though... Hopefully they can just extend WebSocket?

samehhady, 2015-12-18:
actually I've been away for some time as I was traveling, flashing now the ESP with the latest Espruino and trying the ws module

samehhady, 2015-12-21:
@gfwilliams  I am receiving weird error after updating to the latest bins.

Seems related to the socket API, is there any new changes that happened?

```
Uncaught Error: Field or method "on" does not already exist, and can't create it on undefined
 at line 1 col 27
this.socket=b;var a=this;b.on("data",this.parseData.bind(thi...
                          ^
in function called from system
```

samehhady, 2015-12-21:
It doesn't seem related to the module itself as all my old versions that used to work are giving me the same error!

gfwilliams, 2015-12-21:
Thanks - I actually just committed some changed WebSocket code that fixes this (bad timing!).

I also hit that bug just now - there were some recent changes for networking on ESP8266, and honestly it seems to have completely screwed the networking up.

If you use 1v84 then it should all work though. I'll file a bug for this now and hopefully it'll get fixed soon.

gfwilliams, 2015-12-21:
Just to add - not sure what you can do on ESP8266 - I think realistically you'll have to wait for the next build with this fixed.

While I committed a change with the `on` error fixed, there's something deeply wrong with networking now, and it seems to be transmitting the wrong characters!

gfwilliams, 2015-12-21:
I think with the recent update this is fixed now anyway.

Sorry for wasting your time - there was [a post on the forum](http://forum.espruino.com/conversations/279206) with some tweaks for servers, and I thought I'd remove the JSON issue at the same time as pulling them in.

closed 2015-12-21 by gfwilliams

samehhady, 2015-12-21:
great, after pulling updates and doing a new build, all works good now and was able to use the socket client perfectly