espruino/EspruinoDocs #208: MQTT Client: mqttStr not found

andriyadi, 2015-12-31:
I think there's bug on MQTT module.

`mqttStr` is not found when calling connect function. I guess the problem is [here](https://github.com/espruino/EspruinoDocs/blob/ac6b7640a1f3711eba6bdb9774baa0d0c1136925/modules/MQTT.js#L295) 
```
return mqttPacket(cmd, 
           this.mqttStr( this.protocol_name )/*protocol name*/+
           this.protocol_level /*protocol level*/+
           flags+
           keep_alive,
           payload);
```

Should be written `mqttStr`, instead of `this.mqttStr`

Thanks. Happy New Year!

mkarliner, 2016-01-02:
Yep, I've fallen over this one as well.

gfwilliams, 2016-01-04:
Sorry about this - I'll get the website updated with the new version later today.

commit 3295645e (2016-01-04): Fix regression caused by MQTT minification tweaks (fix #208)
closed 2016-01-04 by gfwilliams