espruino/EspruinoDocs #220: boards/MicroBit.md

Jackbennett, 2016-01-29:
On the pinout diagram. pins D4-12 list LED Metrix column 1-9 and D12-15 list row 1-3. Just seems odd for a 5x5 grid. Obviously I don't have board yet to cross check.

How are you making those diagrams by the way they're nice!

gfwilliams, 2016-01-29:
Yep, it's right. its actually wired as a 9x3 matrix but laid out physically as a 5x5 - just to make it as difficult as possible for any kids to actually program it at a low level :)

I don't really know why it was - especially as `9+3=12` but `5+5=10`. The only thing I can think of is it reduces flicker if it's only scanning out 3 vs 5 rows.

The diagrams are just made by `scripts/build_board_docs.py` in the main Espruino project - it builds them from a description file like this: https://github.com/espruino/Espruino/blob/master/boards/MICROBIT.py

... it's a bit rough, but it ensures that the docs and the actual binary are in sync

closed 2016-01-29 by gfwilliams

Jackbennett, 2016-01-29:
How strange. Might be some component count our layout/trace reason. Just itching to get a hold of one. 

The diagrams work great, I wondered if svg might be a better fit but can't think of a reason. Thank you

gfwilliams, 2016-01-29:
Yes, ideally the board description files would skip the CSS and would just have start/end positions for each row of pins - then different types of output could be made.

It's just one of those things that got hacked up early on though - and now there are so many boards it's a right pain to change :)