espruino/EspruinoDocs #233: devices/Bluetooth.md

afitterling, 2016-04-15:
I have a command interpreter for communication between two espruinos via the HC-05 I send you the code recently. May we add it?
https://github.com/afitterling/espruino

k r Alex

gfwilliams, 2019-02-21:
Sorry for the crazily late reply to this,

Thanks! I think as an example (so a JS file in https://github.com/espruino/EspruinoDocs/tree/master/examples with a bit of explanation in a comment at the top) it might make a lot of sense?

afitterling, 2019-02-23:
Hi, no prob. I have to admit, the code I posted is tight enough but needs refactorings. I also have to admit, I switched to particle (particle.io) devices already. So not sure, wether I find time to do this. warm regards, Alex

afitterling, 2019-02-23:
(not because of the late reply, rather because to simply focus on iot products)

closed 2020-11-05 by gfwilliams