espruino/EspruinoDocs #240: How to `load()` after `reset()`?

ceremcem, 2016-04-29:
I need to reboot if some kind of heartbeat is overflowed, but `reset()` causes a reboot without app code `load()`ed. 

How can I simulate a power toggle with a command?

gfwilliams, 2016-04-29:
Just type `load()` - that's what it does.

Please can you ask these questions [on the forum](http://forum.espruino.com/) - that's what it's for - this isn't really an issue,

closed 2016-04-29 by gfwilliams

ceremcem, 2016-04-29:
We were expecting a `reboot()` or `restart()` like command. You may consider a section with these keywords in the documentation. 

Moreover, this might be considered as a (small) bug, since this is an expected command and we are unable to find it in the documentation. Interaction should not be required for such a small need, IMO

trandi, 2016-04-29:
Really !? A bug because it's not what YOU expected and YOU could not find
it ???
On a project where Gordon (and others) has invested soooooo much time FOR
FREE ??!!

Dan

gfwilliams, 2016-04-29:
Well, [the docs for reset](http://www.espruino.com/Reference#l__global_reset) do link to `load`, and the docs for `load` say:

> Espruino will resume from where it was when you last typed save().

Its seems relatively clear?

gfwilliams, 2016-04-29:
I'll modify the docs for 'load' though to mention restart/reboot, in case someone searches for it.

ceremcem, 2016-04-29:
Thank you. I've forked the documentation, I'll create pull requests for similar needs next time.