espruino/EspruinoDocs #248: HTU21D: issue with degree symbol (°)

The-Alchemist, 2016-05-22:
I get a weird symbol after copy/pasting the code from http://www.espruino.com/HTU21D:

[image]

I copy/pasted from Chrome.  Anyone else see this problem?

MaBecker, 2016-05-23:
This is not a module issue.

Typing direct to console works

````
t = 23;
=23
>console.log("t = " + t + " °C");
t = 23 °C
=undefined
> 
````

using upload it is printing more than °C 

````
t = 23;
console.log("t = " + t + " °C");

/* output

t = 23 Â°C
=undefined

*/
````

Plattform: Espruino version 1v85 on ESP8266

gfwilliams, 2016-05-23:
It's actually a Web IDE issue (kind of) - I think the degree symbol is being treated as utf-8 unicode, so is actually being sent as 2 characters - and the Web IDE isn't decoding the received characters back into one character.

A quick fix would be to make the tutorial use the correct numeric ascii code for a degree symbol...

The-Alchemist, 2016-05-23:
Thanks, guys!

@gfwilliams : If you point me in the right direction in the Web IDE code, I might be able to fix the issue.

MaBecker, 2017-02-02:
Hi @The-Alchemist,

does it work for you too ?

[image]

gfwilliams, 2017-02-02:
Thanks for spotting this - I think this was fixed a month or so ago.

Sorry I missed the previous post, but FYI it's in `unicode.js` I think

closed 2017-06-15 by gfwilliams