espruino/EspruinoDocs #267: boards/MicroBit.md BTN1 shown on wrong pin

veryalien, 2016-08-14:
BTN1 is on pin d5, not on pin d4 as shown on the Microbit edge connector pinout.

veryalien, 2016-08-14:
It seems to be a problem with the Microbit edge connector itself, not with your diagram

veryalien, 2016-08-14:
After checking and re-checking it is a problem: Espruino is consistently using the 'wrong' pin numbers. Unfortunately, the pin numbers don't match those given here [https://www.microbit.co.uk/device/pins] and the Microbit edge connector.

gfwilliams, 2016-08-14:
Not sure I understand - D0/1/2/etc are all in the right place and work fine? I'm sure I've used them in the past?

Or are you using integers like `2` instead of `D2`?

gfwilliams, 2016-08-14:
Oh right. you opened a new bug as well: https://github.com/espruino/Espruino/issues/896

gfwilliams, 2016-08-15:
This'll be fixed automatically on the next website update now.

closed 2016-08-15 by gfwilliams