espruino/EspruinoDocs #268: Websocket client needs masking

gfwilliams, 2016-08-22:
`ws` on node.hs is fine with it, but vertx.io drops connections if it's not there:

http://forum.espruino.com/conversations/291544/#comment13167995

The code that needs changing is in `send`, in: https://github.com/espruino/EspruinoDocs/blob/master/modules/ws.js#L175

There is already some unmasking code, so you're basically just doing the same thing again: https://github.com/espruino/EspruinoDocs/blob/master/modules/ws.js#L154

injecteer, 2016-08-22:
I fixed the masking issue.
Now it's active by default and can be deactivated via the `masking` option.

I also fixed the ping sending. The interval-timer is cleared, if the websocket is disconnected.

[ws.zip](https://github.com/espruino/EspruinoDocs/files/430512/ws.zip)

gfwilliams, 2016-08-22:
Thanks! I've just updated - hope that works for you!

closed 2016-08-22 by gfwilliams

injecteer, 2016-08-24:
I tested that for sending and receiving, and it's working like charm. Thanks!

commit 25c74fa2 (2016-08-22): Websocket fixes. Injecteer's changes (masking,killing interval on socket close), plus whitespace, receive >125 chars, and receiving multiple messages at once. (fix #268)