espruino/EspruinoDocs #273: SSD1606 - has some default POR power on reset values

programmicha, 2016-08-30:
SSD1606 - has some default POR power on reset values, so it would be nice to send the commands for those default values only if needed

gfwilliams, 2016-09-05:
Apart from a very slightly faster power up time, does this gain anything? It would almost certainly use up more code on Espruino, which is probably what people care more about?

programmicha, 2016-09-05:
it needs an optional config parameter, which is checked

```
  if(config.setPor){
    this.setPor = true;
  }

```
and used

```
  if(this.setPor) {
    this.por();
  }
```

to control this

```
SSD1606.prototype.por = function() {
  this.scd(0x01, 0xB3);
  this.scd(0x03, 0xEA);
  this.scd(0x04, 0x0A);
  this.scd(0x3A, 0x04);
  this.scd(0x3B, 0x08);
  this.scd(0x2C, 0xA0); 
  this.scd(0x3C, 0x73);
  this.scd(0x11, 0x03);
  this.scd(0x4E, 0x00);
  this.scd(0x4F, 0x00);
  this.scd(0xF0, 0x1F);
};
```

where each scd stands for

```
  digitalWrite(this.dcPin, LOW);
  this.spi.write(command, this.cs1Pin);
  digitalWrite(this.dcPin, HIGH);
  this.spi.write(data, this.cs1Pin);
```

for me it looked like a good trade, but feel free to close the issue :-)

ps: is using shorthand style for conditions better for memory?

gfwilliams, 2016-09-06:
Ahh, that's great. Sorry, I assumed you wanted to do something a bit like:

```
if (options.x!=default_xvalue) this.scd(0xF0, options.x);
if (options.y!=default_yvalue) this.scd(0xF0, options.y);
// ...
```

Which (IMO) would have been overkill and a bit of a source of bugs :)

commit 9c14c544 (2017-06-08): add function to set registers back to power-on values. fix #273
closed 2017-06-08 by gfwilliams