espruino/EspruinoDocs #288: boards/WiFi.md

sammachin, 2016-11-12:
The Chrome IDE app's instructions for updating the Espriuno WiFi are a little confusing, it says to press BTN1 and Reset but the WiFI board only has one button (I assume BTN0) and there is no reset button.

I managed to work out that holding down that button and shorting the RST pad to GND put it into update mode but it would be handing to add some notes to the WiFi board page explaining this.

gfwilliams, 2016-11-12:
Thanks - I'll try and get that fixed.

You actually only need to hold the button down while plugging into USB (like the Pico).

sammachin, 2016-11-12:
ah ok, I did try that but wasn't getting any joy, its a but fiddly given
the location of the button and the USB port!

Needs a jumper wire but the reset tab is fairly easy to do then

gfwilliams, 2017-06-08:
I just checked this, and don't see `reset` anywhere in the instructions.

All I can think is you might have suggested the wrong board (Espruino Original) in the flashing options, and so it gave you the instructions for that. Is that possible?

gfwilliams, 2017-06-15:
Any update? or shall I close this?

closed 2018-01-31 by gfwilliams