espruino/EspruinoDocs #312: puck/Web Bluetooth On Linux.md

headdab, 2017-01-01:
Compile didn't work for me.  Followed your step exactly on an ubuntu 16.04 system and got:

  CC       gdbus/mainloop.lo
In file included from /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h:9:0,
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from gdbus/mainloop.c:28:
/usr/include/glib-2.0/glib/gtypes.h: In function '_GLIB_CHECKED_ADD_U64':
/usr/include/glib-2.0/glib/gmacros.h:232:53: error: size of array '_GStaticAssertCompileTimeAssertion_0' is negative
 #define G_STATIC_ASSERT(expr) typedef char G_PASTE (_GStaticAssertCompileTimeAssertion_, __COUNTER__)[(expr) ? 1 : -1] G_GNUC_UNUSED
                                                     ^
/usr/include/glib-2.0/glib/gmacros.h:229:47: note: in definition of macro 'G_PASTE_ARGS'
 #define G_PASTE_ARGS(identifier1,identifier2) identifier1 ## identifier2
                                               ^
/usr/include/glib-2.0/glib/gmacros.h:232:44: note: in expansion of macro 'G_PASTE'
 #define G_STATIC_ASSERT(expr) typedef char G_PASTE (_GStaticAssertCompileTimeAssertion_, __COUNTER__)[(expr) ? 1 : -1] G_GNUC_UNUSED
                                            ^
/usr/include/glib-2.0/glib/gtypes.h:422:3: note: in expansion of macro 'G_STATIC_ASSERT'
   G_STATIC_ASSERT(sizeof (unsigned long long) == sizeof (guint64));
   ^
Makefile:5559: recipe for target 'gdbus/mainloop.lo' failed
make[1]: *** [gdbus/mainloop.lo] Error 1
Makefile:3092: recipe for target 'all' failed
make: *** [all] Error 2
Exit 2

when I ran make.  Ideas?

Thanks.

gfwilliams, 2017-01-04:
Strange - it seems others have been able to get this working fine following the steps, and I've done it on 2 PCs. Could there be something different about yours? Was it updated from an earlier version, or could you have updated glib somehow?

gfwilliams, 2017-01-23:
Any news on this? Otherwise I'll close it.

headdab, 2017-01-23:
I haven't tried it again.  Give me a day or two and I'll try it again.
There's nothing really special about my machine configuration.

gfwilliams, 2017-06-08:
Any news, or should I close this?

headdab, 2017-06-08:
I don't think I'm going to have enough time to debug this in the near
future.  Go ahead and close it.
Thanks!

gfwilliams, 2017-06-09:
Ok, thanks! New releases of Ubuntu have a new enough bluez in them - so very soon pretty much everyone will have something that 'just works'

closed 2017-06-09 by gfwilliams