espruino/EspruinoDocs #316: boards/Puck.js.md

basisdude65, 2017-01-13:
I am a newbie and would like to provide feedback using iOS based tools which as per usual differ from the same product for Android.  I have managed to upload the firmware using the recommended tools family yet the UI is very different in iOS.  
1) iOS  you must use the "nRF Toolbox" as the "nRF Connect" does NOT contain the DFU option
2) iOS "nRF Connect" tool has some little quirks that are worth mentioning in greater detail.

Please advise how to best share the information and screen captures I now have.  I guess perhaps a video is in order yet I will admit I have never created a proper video for such things. 

Marc Nevin  aka the newbie

gfwilliams, 2017-01-13:
Hi,

Thanks! It does say in the docs that you can't use that on iOS, but I think people are drawn to the video :)

A video would be cool, but it's a bit of work - it might be better for you to just edit the text that's on the website for a start?

You should be able to go here:  https://github.com/espruino/EspruinoDocs/blob/master/boards/Puck.js.md#via-nrf-toolbox-app-android--ios 

And if you make a GitHub account you should be able to edit it online and then send the changes back to me. There'll be some tutorials about doing that online if you need some help.

Same applies to the Quick Start guide - which I guess is where you encountered the differences with the nRF connect app? https://github.com/espruino/EspruinoDocs/blob/master/puck/Puck.js%20Quick%20Start.md

gfwilliams, 2017-06-15:
Do you think you'll do this? Or should I do a video?

closed 2018-01-31 by gfwilliams