espruino/EspruinoDocs #318: puck/Web Bluetooth On Linux.md

jjok, 2017-01-21:
I spent ages trying to build the latest Bluez on Ubuntu 16.10, but couldn't get the new versiont to start (or the old version to stop).

It's actually a lot easier to get Web Bluetooth working. You just need to edit the Bluetooth config and add the experimental flag there, building from source.

```
sudo nano /lib/systemd/system/bluetooth.service
```

Update the line

```
ExecStart=/usr/lib/bluetooth/bluetoothd
```
so it reads

```
ExecStart=/usr/lib/bluetooth/bluetoothd --experimental
```

Save and exit.

```
sudo systemctl daemon-reload
sudo systemctl restart bluetooth
```

From this tutorial 
https://learn.adafruit.com/install-bluez-on-the-raspberry-pi/installation

gfwilliams, 2017-01-23:
I think on Ubuntu 16.10 you already have the up to date version of bluez, which is why this works? https://launchpad.net/ubuntu/yakkety/amd64/bluez

Thanks for the info though - I'll add it to the beginning of the page. It's a lot easier than the recompile if you have a newer Ubuntu :)

jjok, 2017-01-23:
Yeah. Ubuntu 16.10 comes with Bluez 5.41 which is enough to get Web Bluetooth working.
Your docs suggest that Bluez 5.43 is required, which is why I was try to compile it. If you could explain that a bit more clearly it would be great.

Thanks.

gfwilliams, 2017-01-23:
Hopefully that makes it a bit clearer? My understanding is that 5.43 'just works', and 5.41 needs the experimental flags on it - could be wrong though!

closed 2017-01-23 by gfwilliams

jjok, 2017-01-23:
Ah, right. Well, it certainly turned out to be a lot simpler to just update the flag, rather than rebuilding. I just couldn't get the old service to stop so I could start the new one.

jjok, 2017-01-23:
I think that change makes it a lot more clear. Thanks.

commit 9febdb29 (2017-01-23): update docs for Ubuntu >16.04 - fix #318