espruino/EspruinoDocs #340: Graphics library independent display drivers

opichals, 2017-03-13:
I have been playing with [SSD1306 OLED display](https://www.espruino.com/SSD1306) on an Esp8266 board which doesn't have the Graphics compiled-in by default and therefore not directly possible. However it is unfortunate as I was able to easily implement setPixel on top of a manually created ArrayBuffer to make this work for rudimentary approach.

It would nice for the display drivers didn't depend on the Graphics library directly so that one would be able to connect any device and handle the framebuffer operations on their own while using the drivers' i2c/spi related code. I can provide my SDD1306 as an example.

gfwilliams, 2017-03-13:
Surely it's better to focus our efforts on adding the graphics library to the ESP8266 build instead? I mean, worst case it could implement `setPixel` and nothing else.

You can always just define `Graphics.createArrayBuffer` yourself in JS and then you can use the libraries as-is?

opichals, 2017-03-13:
Ahh, you are right! Fair enough, I could have just defined Graphics.createArrayBuffer.

I guess then it boils down to whether the Graphics lib should be included or not. I mean I assume it would be quite a frequent use-case to have a device not having any display and therefore waste a lot of space. But I must agree that a minimal setPixel()-only Graphics should not waste too much really and would be of great help.

gfwilliams, 2017-03-13:
There's a bug open for creating an ESP8266 image with Graphics: https://github.com/espruino/Espruino/issues/1110

Realistically the only boards that couldn't run that image are the ESP01 boards - which don't really have enough IO to connect a display anyway...

closed 2017-03-13 by gfwilliams