espruino/EspruinoDocs #356: DS3231 module, setDate throws error

ancienthero, 2017-05-17:
date setting throws an error

```
>rtc.setDate(17,05,2017);
Uncaught Error: Field or method "DST" does not already exist, and can't create it on undefined
 at line 1 col 13
this.options.DST?3===b&&7===d&&23<a?!0:10===b&&7===d&&23<a?!...
            ^
in function "h" called from line 1 col 146
...6)]);this.dstStatus=h(a,b,d)
                              ^
in function "setDate" called from line 1 col 23
rtc.setDate(17,05,2017);
                      ^
>console.log(rtc.readDateTime());
```
but date has been set though 
```
>console.log(rtc.readDateTime());
17/05/17 00:36:16
```

gfwilliams, 2017-05-22:
Thanks - I believe I have fixed it now. I don't have a DS3231 here to test though, so please could you re-upload and see if it still happens?

closed 2017-05-22 by gfwilliams

ancienthero, 2017-05-22:
tested, it's ok now!

commit 4377e28e (2017-05-22): fix #356