espruino/EspruinoDocs #364: Even though `ws` module exists, IDE still tries to fetch `ws` from NPM.

trusktr, 2017-06-02:
I gave the NPM beta setting a shot, and when I `require('ws')` it fails with those yellow toast messages about not finding modules or something, but `ws` is an Espruino module, so should it find that one first and just work?

gfwilliams, 2017-06-02:
I think the issue you're seeing is this one? https://github.com/espruino/EspruinoWebIDE/issues/106

I'm afraid the NPM support in the IDE really isn't usable at the moment. Not to mention that fact that the majority of NPM modules pull in so much extra code they use up all your RAM before you've written any code :(

trusktr, 2017-06-02:
Hehe, yeah, I thought the same about some packages having so many modules.

`ddp.js` seems like a small one, it only depends on `wolfy87-eventemitter` which has no other dependencies, but that one failed with a bunch of those yellow toasts as well.

Maybe I can just compile an entry point myself with webpack or rollup and dead-code elimination plus tree shaking. I'm coding in vim anyways, then pasting into the IDE, but I could also just upload code (f.e. a bundle) with the CLI.

trusktr, 2017-06-02:
Maybe that could be a strategy for NPM support: treat the IDE file as entry point, scan for dependencies like you already do, `npm install` them somewhere, then bundle the entry point with an existing tool, and upload that bundle.

With this technique I know at least there wouldn't be all these "not found" errors I'm seeing.

I'm just guessing, but I think you're fetching NPM packages in a custom way? Otherwise `npm` wouldn't fail the way the IDE is failing.

trusktr, 2017-06-02:
Babel also has custom presets, so if we could list what language features Espruino has or doesn't have, and also caveats like function hoisting, then we could make a Babel preset and/or webpack loader that can always output code that espruino will understand, even if the entry-point is written with function hoisting.

trusktr, 2017-06-02:
Cool, even the [`asteroid`](https://www.npmjs.com/package/asteroid) package is fairly light, the dep tree is

```
asteroid
  ddp.js
    wolfy87-eventemitter
  wolfy87-eventemitter
  lodash-assign
```

which comes comes down to just the `asteroid, ddp.js, wolfy87-eventemitter, lodash-assign` packages.

I'd like to try and see if it's usable, I'll get back with webpack/rollup results...

gfwilliams, 2017-06-05:
Yes, the best bet is to try with the CLI.

The NPM support is custom - because it would be super painful to try and run NPM in the web browser. I think I should probably just remove that option - it's caused far more harm than good.

gfwilliams, 2017-06-08:
I've just removed the NPM option - so next update we shouldn't have this issue

closed 2017-06-08 by gfwilliams

trusktr, 2017-06-08:
I think it's worth a shot solving this with an established tool like rollup or webpack.

Rollup in particular had minimal clean output: https://rollupjs.org

trusktr, 2017-06-08:
Not sure what you already were doing, but you can replace require calls to Espruino modules with something else so rollup doesn't try to get those modules from NPM.

Another option would be to store modules under the @espruino org, so it is easy to sketchy which module:

```js
const ws1 = require('ws')
const ws2 = require('@espruino/ws')
const MPU = require('@espruino/MPU6050')
```

gfwilliams, 2017-06-08:
Does it work in the browser? I don't think it does...

Right now, you can use rollup and then the CLI to upload the finished file on the command-line, which will work great.

And to be honest most people that want this are the same people that are used to Node.js dev and are using their own editors and not the Web IDE, so I don't think it's worth me adding to the IDE.

Look at the NPM support I just removed. Even when it was working fine, nobody actually cared. I spent ages implementing it, and all I heard back was what I knew I would "I tried to require module X and it didn't fit", plus all the issues people encountered from enabling it and using the wrong modules by accident. I think if I spend another few weeks on this I'm likely to get a very similar result.

If you care about it, you can try adding it yourself if you want? I think realistically it'd probably be more useful to add as an extension to EspruinoTools though so it works in the CLI - and then if you *really* want the IDE as well you can use the Web IDE server in EspruinWebIDE