espruino/EspruinoDocs #365: [BUG?]  Sometimes a trailing period appears on floating point values, which causes JSON.parse to fail.

trusktr, 2017-06-03:
Hey hey, I'm using the MPU6050 module, and I'm logging values to the console. Sometimes when I use `mpu.getGravity()` I'll get an array that when printed to console shows something like

```
-2., 3, 4
```

where one of the numbers may have a trailing period. I would assume that the number was meant to be `-2.0`, which should actually print to just `-2`, and I'm not sure why this happens. It isn't a string, because otherwise the output would include quotes, like

```
"-2.", "3", "4"
```

So somehow numbers are bring stringified to contain trailing periods.

When I use `JSON.stringify()` on the array, the result also includes the trailing period, for example the result is a string like

```
"[-2., 3, 4]"
```

Then, when I send this from espruino to computer, and eventually to a browser, and call `JSON.parse()` on it, I get an error like

```
(unknown) Uncaught SyntaxError: Unexpected token , in JSON at position 3
```

In JavaScript (like Node.js or Browsers) the numbers are never stringified with a trailing period, and so JSON.parse always works in this manner.

As a workaround for the time being, I can use a regexp to detect the `.,` and fix it on the client.

gfwilliams, 2017-06-05:
Thanks - that's pretty frustrating because normal JS code seems file with a trailing `.`, so it's presumably some more strict part of the JSON spec.

trusktr, 2017-06-08:
It fails for me in modern browsers, and also Node:

```
┌─[22:00:33/localhost/chronos/.../src/trusktr+trusktr.io]
└─╼ node -v
v6.9.4

┌─[22:00:35/localhost/chronos/.../src/trusktr+trusktr.io]
└─╼ node
> JSON.parse("{n:2.}")
SyntaxError: Unexpected token n in JSON at position 1
```

 Which "normal JS code" does it work with? In which engine?

trusktr, 2017-06-08:
Oops, my Node example failed because wrong reason, here's actual failure:

```
┌─[22:00:47/localhost/chronos/.../src/trusktr+trusktr.io]
└─╼ node
> JSON.parse('{"n":2.}')
SyntaxError: Unexpected token } in JSON at position 7
```

gfwilliams, 2017-06-08:
I mean `eval("1.")` works fine - it's a valid number in JS. Ideally JSON is a subset of JS, but in this case the extra strictness really messes you up if you're trying to use valid JS.

Hopefully this'll be an easy fix though.

gfwilliams, 2017-06-08:
fixed with https://github.com/espruino/Espruino/commit/6d749d19ba9c959c1c0928a13e9064de0caa5a88

closed 2017-06-08 by gfwilliams

trusktr, 2017-06-08:
> I mean eval("1.") works fine

Ah, I get ya. Yep I thought the same thing. It *should* just work!