espruino/EspruinoDocs #366: [BUG?] [MPU6050] "FIFO overflow!" when using the API as described.

trusktr, 2017-06-04:
I was trying to do this on the Espruino WiFi with the MPU6050 chip:

```js
I2C1.setup({scl:B6, sda:B7, bitrate:100000});
const MPU = require("MPU6050").connect(I2C1);
// 2nd parameter is the fifoRate. The DMP output frequency = 200Hz / (1 + fifoRate)
const DMP = require("MPU6050_DMP").create(MPU, 3);

var data = DMP.getData();
if(data !== undefined) console.log(DMP.getYawPitchRoll(data));
```

but the `DMP.getData();` call causes the espruino to log "FIFO overflow!", and it doesn't seem to work.

I haven't tried it with the interrupt yet, maybe I need to do it that way?

trusktr, 2017-06-04:
Oh, maybe I need the bitrate param?

trusktr, 2017-06-04:
If I follow the example more closely, with the bitrate param and using the interrupt with setWatch, I seem to be getting some results.

However, it seems to render the HTTP server completely unresponsive for some reason.

closed 2017-06-04 by trusktr

gfwilliams, 2017-06-05:
You'd have to post the full example to properly debug this. If your code ends up in some kind of loop so it never actually exits a function then the HTTP server would stop.

But please can you post actual issues like that on the forums instead? I'm trying to keep GitHub for clear bugs with a step to reproduce - otherwise GitHub gets full of 'it doesn't work' posts that nobody looks at except for me.

trusktr, 2017-06-08:
No prob, moving there: http://forum.espruino.com/conversations/306036/

gfwilliams, 2017-06-08:
Perhaps the driver shouldn't be complaining about this so much. If you don't read out the data fast enough, the FIFO will overflow - it's to be expected really. I guess all it means is you lose readings