espruino/EspruinoDocs #367: When I setWatch on B5 for MPU6050 module, HTTP server becomes completely unresponsive, and Web IDE can no longer send code.

trusktr, 2017-06-04:
However, the setWatch loop can continue to log to the console despite both HTTP being unreachable and Web IDE not being able to upload new code (it just hangs).

I've got some code like follows:

```js
function getMPU() {
    I2C1.setup({scl:B6,sda:B7});

    var MPU6050 = require("MPU6050");
    console.log(MPU6050);

    var mpu = MPU6050.connect(I2C1);

    return mpu
}

function connectToWifi(wifiName, options) {
    var resolve, reject
    var promise = new Promise(function(res, rej) {resolve = res; reject = rej})

    var wifi = require("EspruinoWiFi");

    wifi.connect(wifiName, options, function(err) {
        if (err) reject(err);
        resolve();
    });

    return promise
}

connectToWifi("CenturyLink6344", { password : "zpx2psajg7xqat" })
.then(function() {
    const mpu = getMPU();
    const dmp = require('MPU6050_DMP').create(mpu, 3);
    const http = require("http");

    let dmpYawPitchRoll = null

    function dmpLoop() {
        const dmpData = dmp.getData()
        if (dmpData !== undefined) dmpYawPitchRoll = dmp.getYawPitchRoll(dmpData)
    }

    setWatch(dmpLoop, B5, { repeat:true, edge:'rising' });

    http.createServer(function (req, res) {
        res.writeHead(200);
        let result = {
            acceleration: mpu.getAcceleration(), // returns an [x,y,z] array with raw accl. data
            gravity: mpu.getGravity(),  // returns acceleration array in G's
            rotation: mpu.getRotation(), // returns an [x,y,z] array with raw gyro data
            degreesPerSecond: mpu.getDegreesPerSecond(), // returns gyro array in degrees/s
        }
        result = JSON.stringify(result)
        console.log(' ------ dmpYawPitchRoll:', dmpYawPitchRoll)
        res.end(result);
    }).listen(80);

    const wifi = require("EspruinoWiFi");
    wifi.getIP(function(err, info) {
        if (err) throw err

        console.log('IP:', info.ip)
    });
});

function onInit() {
    console.log('Espruino started!');
}
```

The Espruino Wifi server doesn't work when I do that, it is completely unresponsive to the requests sent by my computer.

If I add a console.log, like

```js
        if (dmpData !== undefined) console.log(dmpYawPitchRoll = dmp.getYawPitchRoll(dmpData))
```

then the Espruino WiFi continuously outputs data, so it seems to be running.

Maybe the loop is like a real loop, where all else is blocked? How might I make this work so that I can send the data back to computer? Do I need to make requests inside the dmpLoop instead of making an HTTP server?

trusktr, 2017-06-04:
If I try to get the DMP data in the HTTP handler, then at least HTTP is responsive, but the DMP data doesn't work, it just says "FIFO overflow" every time:

```js
function getMPU() {
    I2C1.setup({scl:B6,sda:B7});

    var MPU6050 = require("MPU6050");
    console.log(MPU6050);

    var mpu = MPU6050.connect(I2C1);

    return mpu
}

function connectToWifi(wifiName, options) {
    var resolve, reject
    var promise = new Promise(function(res, rej) {resolve = res; reject = rej})

    var wifi = require("EspruinoWiFi");

    wifi.connect(wifiName, options, function(err) {
        if (err) reject(err);
        resolve();
    });

    return promise
}

connectToWifi("CenturyLink6344", { password : "zpx2psajg7xqat" })
.then(function() {
    const mpu = getMPU();
    const dmp = require('MPU6050_DMP').create(mpu, 3);
    const http = require("http");

    let dmpYawPitchRoll = null

    //function dmpLoop() {
        //const dmpData = dmp.getData()
        //if (dmpData !== undefined) console.log(dmpYawPitchRoll = dmp.getYawPitchRoll(dmpData))
    //}

    //setWatch(dmpLoop, B5, { repeat:true, edge:'rising' });

    http.createServer(function (req, res) {
        res.writeHead(200);
        let result = {
            acceleration: mpu.getAcceleration(), // returns an [x,y,z] array with raw accl. data
            gravity: mpu.getGravity(),  // returns acceleration array in G's
            rotation: mpu.getRotation(), // returns an [x,y,z] array with raw gyro data
            degreesPerSecond: mpu.getDegreesPerSecond(), // returns gyro array in degrees/s
        }
        result = JSON.stringify(result)
        res.end(result);

        const dmpData = dmp.getData()
        if (dmpData !== undefined) dmpYawPitchRoll = dmp.getYawPitchRoll(dmpData)
        console.log(' ------ dmpYawPitchRoll:', dmpYawPitchRoll)
    }).listen(80);

    const wifi = require("EspruinoWiFi");
    wifi.getIP(function(err, info) {
        if (err) throw err

        console.log('IP:', info.ip)
    });
});

function onInit() {
    console.log('Espruino started!');
}
```

trusktr, 2017-06-04:
@trandi Might you know about it? Is there a reason that the setWatch loop makes the HTTP server unresponsive? Do you know if I can just get the data from the MPU6050 module, transform it, and pass it to the `DMP.getYawPitchRoll` method? I see `getYawPitchRoll` looks at qw, qx, qy, qz; are those for a quaternion? I don't see such a method in the MPU module, does it need all the setup that DMP does in order to get the quaternion, or is there some way to get it from the MPU module?

gfwilliams, 2017-06-05:
What have you actually connected B5 to? It seems like if it's the IRQ from the MPU6050 then it could be triggering so often that it's using up all of the board's processing power? You could also see if `E.getErrorFlags()` returned anything on the board.

Actually looking at the code for `MPU6050_DMP`, it's almost certainly causing your problems. It's got a `while` loop in `getData` that just waits without any kind of timeout. That means that if you call it when data isn't ready it'll just block the whole device until it is - which would explain all your problems.

Can you try `setWatch(pidLoop, B5, { repeat:true, edge:'falling' });`? I think the docs may have got the polarity of the IRQ wrong - which would cause it to basically always trigger and always block at that `while` loop. The original author may not have realised during testing as they weren't trying to do anything else at the same time.

But *please* ask questions like this on the forum in future. That's what it's for - you'll get a much better response to any queries, since realistically very few people read GitHub bugs apart from me.

trusktr, 2017-06-08:
Hey, continuing at http://forum.espruino.com/conversations/306036/#comment13678106, I'll just open issues on here if we can confirm there's a bug.

closed 2017-06-08 by trusktr

trandi, 2017-06-08:
@gfwilliams  I'm only doing that  " while (fifoCount < PACKET_SIZE) fifoCount = DMP.getFIFOCount();" that you're worried about *after* testing "if (status & 0x02) { // otherwise, check for DMP data ready interrupt (this should happen frequently) "
I would have imagined that offers a guarantee that that fifoCount will eventually increase very rapidly.

gfwilliams, 2017-06-08:
@trandi thanks for weighing in - I don't suppose you get anything like this happening for you? I'm struggling to see what could be causing it.

Looks like this is getting discussed here now: http://forum.espruino.com/conversations/306037