espruino/EspruinoDocs #369: puck/Puck.js HTTP Proxy.md

mikeboski, 2017-06-21:
when running the EsprinoHub  and using https://www.espruino.com/ide I  run:

NRF.requestDevice({ filters: [{ services: ['1823'] }] }).then(function(device) {
  require("ble_http").httpRequest(device, "pur3.co.uk/hello.txt", function(d) {
    print("GET:",JSON.stringify(d));
  });
});

I get this result: (In https://www.espruino.com/ide console)
Connected
Got service
Got characteristic
Set URI
Got characteristic
Written GET
Disconnected
GET: {"buffer":[72,101,108,108,111,32,87,111,114,108,100,33,10]}

then the EsprinoHub shuts down (goes back to command prompt)
Here is what was in the terminal window(on Rasp Pi 3):
<HTTPProxy> ed:32:bd:69:f7:95 connected (whitelisted: true)
Scanning stopped.
<HTTPProxy> URI -> pur3.co.uk/hello.txt
<HTTPProxy> State => 1
<HTTPProxy> Status code => 200 1
<HTTPProxy> Status code => 200 5
<HTTPProxy> Disconnected
BLE broken? No advertising packets in 10 seconds - restarting!

How do I keep it running?

gfwilliams, 2017-06-21:
Hi - sorry about this. It's odd this didn't get noticed before.

I've fixed it now in the EspruinoHub software, so if you update it from GitHub and restart it should be fixed.

closed 2017-06-21 by gfwilliams

mikeboski, 2017-06-21:
tested and worked.

Thanks

mikeB