espruino/EspruinoDocs #383: help with VL53L0X module "dataView not defined"

ZacharyLightman, 2017-10-01:
i cannon get the VL53L0X sensor to run on the original espruino,  the module may be broken. 
i used the following example code:

I2C2.setup({sda:B3,scl:B10});
digitalWrite(B4,1); // set XSDN -> turn the sensor on
var laser;
setTimeout(function() {
  laser = require("VL53L0X").connect(I2C2);
  // make one reading
  console.log(laser.performSingleMeasurement().distance+" mm");
  // or repeatedly measure
  setInterval(function() {
    // sets LED brightness depending on distance
    var dist = laser.performSingleMeasurement().distance;
    analogWrite(LED1, (300-dist)/300, { soft:true, freq:200 });
  }, 50);
});

The errors is get are:

Uncaught ReferenceError: "DataView" is not defined
 at line 1 col 201
..._STATUS,1)[0]&1);var a=new DataView(this.r(20,12).buffer);va...
                              ^
in function "performSingleMeasurement" called from line 3 col 46
...r.performSingleMeasurement().distance+" mm");
                              ^
in function called from system

gfwilliams, 2017-10-02:
Which version of the Espruino firmware are you using, and on which device?  DataView was introduced a few months ago, so if your version of Espruino is older than that it would produce that kind of error.

ZacharyLightman, 2017-10-02:
Oh, I didn't realize! When I get back home I will check. Thanks!

closed 2017-10-18 by gfwilliams