espruino/EspruinoDocs #401: boards/ESP32.md

johnsd3, 2017-12-15:
To whom it may concern,

I think a note is needed on how to prepare the toolchain for the ESP32 on the espruino/esp32 page.

There's something wrong with the xtensa tools that come from Espressif.  The latest version does not work to build Espruino.  However if you use an older version of the tools it works fine.

This is documented in this thread:
https://forum.pycom.io/topic/1429/solved-having-issues-build-firmware-from-soure/4

Regards,

jumjum123, 2017-12-16:
> build-firmware-from-soure/4

Espressif id working hard and for long time, to get V3.0 up and running.
This includes tons of changes in esp-idf and in toolchain.
In my environment I've Espruino with some first steps to Bluetooth based on
V3.0rc1 and actual toolchain.
I would recommend to wait until V3.0 is released, its only rc1 right now.
Once this happened, a lot of changes in make, sources,documentation etc.
will arrive step by step in ESP32-branch of Espruino.
Other option you have, is to work with released ESP32 version based on
esp-idf v2.1 with older toolchain.
Sorry, but I'm pretty sure, there is nobody spending time to get actual
toolchain running for V2.1

gfwilliams, 2017-12-16:
Under `Build the Firmware` the instructions say to use `source scripts/provision.sh ESP32` which should pull in the correct version of the SDK for what's in the Espruino repo at that time. It seems to work great for me...

Is there somewhere else in the docs that mentions something about the SDK that would be confusing though?

johnsd3, 2017-12-16:
If I install the latest xtensa toolchain prior to the "build the firmware" steps, I get a similar linker error described by another user:
https://github.com/espruino/Espruino/issues/1265

If you install the older toolchain described in the link I previously supplied, everything works fine after then completing the "build the firmware" steps.  

My thought would be to include a preparatory step to get the toolchain setup right.  Fine if you want to wait per jumjum123's request but it took me a few days to get things working correct.

Thanks again.

gfwilliams, 2017-12-18:
What OS are you using? We do Travis builds for ESP32 using nothing except the instructions you saw, on every commit, and they work fine: eg. https://travis-ci.org/espruino/Espruino/jobs/317474595

It's possible that in your case it isn't pulling in the correct toolchain because you have a newer one that's already installed?

johnsd3, 2017-12-18:
Ubuntu.  

I think I see the issue.  It sounds like your provisions.sh script is setting up the toolchain.  I didn't understand that and made the assumption that I needed to install to the toolchain before because that was not discussed.  I guess a note about what provisions.sh does would be helpful.   Thanks again.

Regards,

commit 67e6e6c5 (2017-12-18): fix #401
closed 2017-12-18 by gfwilliams