espruino/EspruinoDocs #404: puck/Puck.js iBeacon.md

ornago, 2018-01-02:
I tried so many ways but it doesn't work. Always get an error like following.

```
Uncaught TypeError: Expecting object, array or undefined, got Array
 at line 1 col 49
NRF.setAdvertising(exports.get(a),{interval:100})
                                                ^
in function "advertise" called from line 6 col 2
});
 ^
```

running 1.95 on my puck
uuid : [0xa9, 0x5e, 0x99, 0xea, 0xd1, 0xa2, 0x45, 0xea, 0xbf, 0x0c, 0xc7, 0xa2, 0xe4, 0xc8, 0x4d, 0x71]
hope thats the right format, also tried decimal ...

ornago, 2018-01-04:
Also, missing semicolon in "ble_ibeacon.js" line 13

gfwilliams, 2018-01-04:
Sorry about this - it looks like a regression in 1v95 that meant it expected an array of arrays when binary data was involved. I've just fixed this - i'll update the modules later today and at that point it should  start working again

commit 680cff2b (2018-01-04): set advertising data in array brackets - fix for regression in 1v95, fix #404
closed 2018-01-04 by gfwilliams