espruino/EspruinoDocs #426: http://www.espruino.com/Puck.js+Eddystone needs an update

UDXS, 2018-04-21:
Recently, Google started to "phase out" goo.gl. This needs to be fixed on the eddystone page.

http://www.espruino.com/Puck.js+Eddystone

gfwilliams, 2018-04-23:
Thanks. God knows why I base anything on Google products any more :(

gfwilliams, 2018-04-24:
What's a sensible URL shortener to use instead now? bit.ly? I believe some other tutorials use it as well.

UDXS, 2018-04-24:
It is fairly popular. They even have a page for former goo.gl users.

MaBecker, 2018-04-24:
What about setting up a own url shortener like with https://shlink.io/

UDXS, 2018-04-24:
This isn't exactly a URL shortener meant for our (in the context of the Espruino site) own links. It's really meant more as an example shortener for users.

commit 7bcf8531 (2018-11-30): fix #426
closed 2018-11-30 by gfwilliams