espruino/EspruinoDocs #427: Use Readme.io to host documentation?

joakim, 2018-04-21:
[Readme.io](https://readme.io/) is a very user friendly documentation platform that's free for open source projects. It too uses GitHub Flavored Markdown, and supports [importing from Markdown files](https://readme.readme.io/docs/importing-documentation), so I thought it might be possible to "port" Espruino's documentation. While Espruino's current documentation pages work just fine, I thought it might be a nice UI upgrade for the docs – especially for navigation.

I set up a free trial account to see whether it would work, and after a few quick tests it looks like it is possible to import this repo's Markdown files, with just a bit of rewriting. There are a few issues, however, that make "porting" to Readme.io difficult:

1. Imports must be done manually. Apparently it used to support automatic import from GitHub repos, but turns out they have removed that feature.
2. One can only import into the "Documentation" area. Meaning the API Reference entries would also end up in there, not in the "Reference Docs" area where they would belong. ("Reference Docs" has a special layout and navigation tailored for API documentation.) It _is_ possible to manually move each API entry after each import, but I wouldn't enjoy doing that.
3. Each import creates a new version of the documentation. There is a workaround where you delete the old version and rename the new version into the old, but again, that's an extra manual step.

So, unless Readme.io fixes those issues, it looks to me like too much hassle "pushing" Markdown to their service. If only they supported automatic import from GitHub repos, with all content going to their respective areas. I have asked them whether it's on their roadmap..

Then again, maybe you're not interested in using Readme.io for other reasons? Suppose I could have asked first :)

joakim, 2018-04-21:
Some neat features they have:
- Nice landing page
- Custom pages
- StackOverflow type questions and answers
- "Suggest edits" functionality (although that would duplicate this issue queue)
- Announcements
- Ability to customise appearance and turn on/off functionality

gfwilliams, 2018-04-23:
Thanks for the suggestion, but I'm unsure about moving everything to a third-party service - especially as I'd be effectively dropping all the history/PR functionality of GitHub.

Personally I'd have thought the whole look & feel / landing page bit would be far easier for me to change on espruino.com than it'd be to port everything to readme.io?

Also there's all the USES/APPEND_USES stuff which auto-links pages together (and also the 'Examples' links in the software Reference) which I'd imagine would be very hard to replicate - not to mention the auto-tooltips in the Espruino IDE and stuff like that.

If there's anything you think is difficult or think could be changed on espruino.com then let me know though, as obviously I'd really like to make that as easy to use as possible. For example, maybe there should be a general documentation landing page on espruino.com itself?

The software Reference itself is getting a bit too big/difficult to use as well so I'd be interested in anything that could clean that up.

The whole StackOverflow Q&A thing would be great - there's been some hope that the forum itself might gain that feature soon so I haven't been looking around too hard, but if I get some time I'd be extremely interested in trying to add something like it to espruino.com. It'd really clear up a lot of the threads that are now filled with 'me too' posts at the end.

joakim, 2018-04-23:
Sorry, I didn't explain my thinking clearly enough :)

I didn't mean to suggest _moving_ Espruino's documentation over to Readme.io, but rather use it as a "build target" so to speak, instead of today's HTML files. Same repo, same issue queue and pull request functionality, same Markdown files. But instead of outputting HTML files, it would output Markdown files for Readme.io to build its documentation pages from. That way, you'd not be locked to Readme's platform.

The idea was that every update to the repo would automatically re-generate the documentation on Readme.io. That's currently not possible though, but it would be if they added back GitHub integration and improved their import functionality, something they've hinted at might happening. So for now, I don't think it's worth it anyway, but thought I'd share my little investigation anyway.

The reason I looked into this is because I found it a bit tedious to navigate the API reference, and wished that all of the documentation were as easy to navigate as Readme's. But I think you're right, it's probably easier to improve the current documentation, and one could always steal/borrow ideas from Readme and others :)

Namely:
- Landing page, like you mention – would be nice to get an overview of the documentation before diving in
- Sticky left sidebar – with an overview of the section you're currently in, even when viewing a sub-page
- Sticky right sidebar – with a ToC on individual pages

That's pretty much a static landing page + some HTML and CSS changes. I happen to be a web developer, so I could give it a go when I get the time some day.

It would be great if the forum could provide Q&A! That's the functionality of Readme.io that I thought might interest you :)

All in all, I shouldn't criticise, but commend you for the comprehensive and really good documentation! It's only that I wished the layout/design was as good as the content, it deserves that 👍

gfwilliams, 2018-04-23:
Thanks! yes, I'm not a web developer. I do what I can but at the end of the day I'm juggling a lot of stuff, and presentation/ease of use on the website has been a bit neglected recently.

There's actually another selfish reason I'd like to keep people going to espruino.com - it pushes it (and hence Espruino) up the search engine rankings, but it also allows me to get a better idea about what tutorials/documents are of most interest to users and also where they get linked from.

But yes, what you're suggesting with the sidebars sounds great, and should be reasonably easy to do with pretty minor changes to the site. If you're interested it'd be great if you could give it a go. While the site itself isn't open source, the HTML transformation code is, and you could always just tell me what needs tweaking (if anything) in the site template.

However: looking at the links you posted the ones where the sidebar actually works well are relatively small. When everything gets bigger, I think the sidebar becomes far less useful. Your ignite link is a good one: https://apacheignite.readme.io/docs - There's so much stuff in there I wonder if the sidebar really adds much.

I could totally see the left-hand sidebar working for a simple 'getting started' section, but it's hard to see how I could link *all* of Espruino's docs without it turning into a mess.

Same for Espruino - how would we turn the contents of http://www.espruino.com/Tutorials into something that could be put into a sidebar? Apart from the tasks/peripherals headings maybe. I did have a sidebar in the original Espruino website but I actually took it out in the redesign because it just didn't seem to get used.

There's also a big question of making something that can actually be maintained. If there's a big list of links then it might be great for a few weeks, but it'll get out of date pretty quick - so ideally we could come up with something that maybe used the same APPEND_KEYWORDS style approach to ensure that the sidebar always stayed up to date... 

For example, for tutorials the sidebar could actually be a list of 'related' pages - if a tutorial listed 'Internet' as a keyword then other tutorials that also listed 'Internet' could be listed down the side?

wilberforce, 2018-04-24:
Part of the issue I think is the total size of the reference section. One solution would be to break this into sections, or load the content on demand. On the iPad it's not particularly responsive.

gfwilliams, 2018-04-24:
Yes, it could definitely be split up - although I have a feeling that on modern browsers just having one file, but hiding the bits you're not interested in with JS would actually be just as quick, if not more so.

jjok, 2019-01-16:
+1 for splitting that reference page up, or hiding stuff. I always try to use the scrollbar to scroll the page up or down a little, but moving the bar even a tiny bit causes the page to scroll for miles.

gfwilliams, 2024-02-09:
Closing now - we've got device-specific reference pages like https://www.espruino.com/ReferencePUCKJS which help somewhat, and the CSS/JS has been tweaked to make the pages render more quickly recently too.

While splitting up could be useful, I tend to use whole-page-search quite a lot to find stuff, and splitting would break that.

Also I note that right now readme.io is broken, so maybe wouldn't be a great choice: https://trello.readme.io/inactive

If someone wants to do a PR with some changes to improve things I'm all for it but right now I think it's 'good enough'

closed 2024-02-09 by gfwilliams