espruino/EspruinoDocs #429: Espruino Wifi (Wifi module) and docs don't match up.

dweremeichik, 2018-04-28:
According to a require on the Espruino Wifi we see:
```
>require('Wifi');
={
  "connect": function (a,b,d) { ... },
  "disconnect": function (a) { ... },
  "startAP": function (a,b,d) { ... },
  "stopAP": function (a) { ... },
  "scan": function (a) { ... },
  "getIP": function (a) { ... },
  "setIP": function (a,b) { ... },
  "getAPIP": function (a) { ... },
  "setAPIP": function (a,b) { ... },
  "setHostname": function (a,b) { ... },
  "ping": function (a,b) { ... },
  "turbo": function (a,b) { ... },
  "debug": function () { ... }
 }
```
The documentation shows:

- Wifi.connect(ssid, options, callback)
- Wifi.disconnect(callback)
- Wifi.getAPDetails(callback)
- Wifi.getAPIP(callback)
- Wifi.getDetails(callback)
- Wifi.getHostByName(hostname, callback)
- Wifi.getHostname(callback)
- Wifi.getIP(callback)
- Wifi.getStatus(callback)
- Wifi.ping(hostname, callback)
- Wifi.restore()
- Wifi.save(what)
- Wifi.scan(callback)
- Wifi.setAPIP(settings, callback)
- Wifi.setConfig(settings)
- Wifi.setHostname(hostname, callback)
- Wifi.setIP(settings, callback)
- Wifi.setSNTP(server, tz_offset)
- Wifi.startAP(ssid, options, callback)
- Wifi.stopAP(callback)
- Wifi.turbo(enable, callback)

If some or all of the missing items are ESP only, they probably ought to have that in bold by the top of the definition.

In particular, I was looking for getStatus and getDetails.

gfwilliams, 2018-04-30:
You mean these docs? http://www.espruino.com/Reference#Wifi

Usually the functions themselves should be marked with a `Note: this is not/only available in ...` note at the end of the description, but it looks like in the case of Wifi they're not. I'll look into this as it seems to be a bug in the doc builder

gfwilliams, 2018-04-30:
Just fixed with https://github.com/espruino/Espruino/commit/24bfd5cede183fc341e2ba488eeff4c5c87c6ed1

I'll update the docs on the site this week

closed 2018-04-30 by gfwilliams

dweremeichik, 2018-05-01:
Awesome, thank you!