espruino/EspruinoDocs #430: Callback form

dweremeichik, 2018-04-29:
The API documentation does not make it clear what form callbacks are expected to be, whether it is NodeJS, or other. This could be super confusing for beginners as in many places, the documentation states that an object will be returned to the callback (not both an error and result).

gfwilliams, 2018-04-30:
Thanks - just fixed the Wifi ones with https://github.com/espruino/Espruino/commit/eaef9c1440a0d634202b61097fc814921a8e6aae

Also discovered a few cases where ESP8266/32 was different, so tweaked those.

If you find anything else that you think is unclear, if you click the `=>` near the function's heading in the reference you'll get taken to the relevant place in Espruino's source where the docs for it are, and you can modify it and issue a pull request.

closed 2018-04-30 by gfwilliams

dweremeichik, 2018-05-01:
Didn't even realize that the `=>` was there. I'll be sure to make use of it if I come across anything again. (really cool project by the way!)