espruino/EspruinoDocs #437: devices/mosfets.md

mfocht001, 2018-05-29:
The article states that the Vgs for the  IRF3708PBF MOSFET is 3.3 but the datasheet says 4.5v.

gfwilliams, 2018-05-30:
Vgs doesn't have a specific number, there's a scale. This is what I see on the datasheet at http://www.irf.com/product-info/datasheets/data/irf3708.pdf:

[image]

So even at 2.7v it can sink 40 amps.

Or is there something I'm missing?

mfocht001, 2018-05-30:
That part was confusing. The article stated Vgs is what you are looking for. When you go to the datasheet, Vgs is says 4.5v. 
I read this at the top but I think maybe I misunderstood.

Benefits Applications
Ultra-Low Gate Impedance  Very Low RDS(on) at 4.5V VGS  Fully Characterized Avalanche Voltage and Current 

Thanks for the response.

From: Gordon Williams 
Sent: Wednesday, May 30, 2018 3:50 AM
To: espruino/EspruinoDocs 
Cc: mfocht001 ; Author 
Subject: Re: [espruino/EspruinoDocs] devices/mosfets.md (#437)

Vgs doesn't have a specific number, there's a scale. This is what I see on the datasheet at http://www.irf.com/product-info/datasheets/data/irf3708.pdf:

So even at 2.7v it can sink 40 amps.

Or is there something I'm missing?

gfwilliams, 2018-05-30:
>  Very Low RDS(on) at 4.5V VGS

Ahh, yes - it's not lying. At 4.5v the resistance is very low (1/2 of the resistance at 3v). But still, at 3v it's still pretty good - more than good enough for most things :)

closed 2018-05-30 by gfwilliams