espruino/EspruinoDocs #440: tutorials/Quick Start BLE.md

scheib, 2018-05-31:
Oh no, pixl.js arrived without a battery, and documentation doesn't state precisely which kind to pop into it.

wilberforce, 2018-05-31:
I believe it's the same as the puck.js - CR2032

gfwilliams, 2018-06-01:
Yep, CR2032. Thanks for the heads up - I'll add that!

gfwilliams, 2018-06-01:
Actually it does seem to say in the third line down on http://www.espruino.com/Quick+Start+BLE#pixljs

Is it unclear that CR2032 is the battery type?

The whole battery thing is annoying. I'd have loved to have shipped them with batteries but shipping rules are a pain, especially if there's more than one in a pack. I could just about get away with it with Puck.js because the whole thing is covered and in a box, but if it was a separate battery I'd probably have issues.

scheib, 2018-06-07:
Don't know how I misread that page /shrug/.  Looks good to me as it is now.

closed 2018-06-07 by scheib