espruino/EspruinoDocs #472: extend SH1106 to allow delta updates

MaBecker, 2019-01-07:
Like to extend device SH1106 to allow delta updates, because sometime a full update i  bad choice.

* use function Graphics.getModified(reset) be used to identify the byte that changed 

* or scan buffer after each change

* add something like [  _updateDirtyBytes](https://github.com/noopkat/oled-js/blob/master/oled.js#L518-L567) form  Suz Hinton alias noopkat   oled-js

Than use g.flip() to update full screen or g.glip(1) for delta updates

**Any hints or recommendations?**

MaBecker, 2019-01-08:
Module SSD1303 use the same way for complete screen updates.

http://www.espruino.com/modules/SSD1306.js

gfwilliams, 2020-11-05:
Closing this for now - don't see much interest and I think given the relatively small buffers it's work just keeping it simple and pushing everything.

closed 2020-11-05 by gfwilliams