espruino/EspruinoDocs #473: Improving the BME680 module

AkosLukacs, 2019-01-10:
Hi!

Thanks for the help in the forums, I started to look at the BME680 module, if I can reduce it's minified size.
I made some progress: wrapped the whole thing in an IIFE, tried to the constants out from the `C` object, but looks like Esprima doesn't eliminate unused ones, they stay there with shortened name. So just commented out the unused constant for now...

Current / original: 9954 bytes minified, memory usage: 774
Modified: 7633 bytes minified, memory usage: 680
Unused constants commented out: 6594 bytes minified, memory usage: 606
Memory usage is on the Pixl.JS with minimal init code

Fixed some bugs: the module is accidentally modifying / leaking a couple of globals: `data`, `factor`, `heater_res_x100`.

Why the sparse arrays for `par_t`, `par_p`, `par_h`, `par_gh`? Looking at the code, `par_t[0]` and others are never accessed? Remnants from some original code?
```javascript
this.cal.par_gh = [,vc.getInt8(GH1_REG)
                   ,vc.getInt16(GH2_LSB_REG, true)
                   ,vc.getInt8(GH3_REG)];
```
So cal.par_gh is `[ undefined, -20, -14548, 18 ]`.

Current progress is [here](https://github.com/espruino/EspruinoDocs/compare/master...AkosLukacs:patch-2#diff-6c0c9fa98ad3e4d1f78f4854d7d20310) sorry, the diff is a mess :/

gfwilliams, 2019-01-10:
Hi,

Thanks for taking a look at this. It's great to fix those accidental leaks - any chance they could be pulled out into a PR of their own before all the size changes get made?

There's something strange going on with the constants. Usually, that pattern (private `C` variable) works great. For example http://www.espruino.com/modules/SSD1306.js uses it and minifies to http://www.espruino.com/modules/SSD1306.min.js without issues. I'd really prefer to figure out why it's not minifying rather than swapping everything back to global variables, which seems like a bit of a step back. If it could be made to properly minify `C` then instantly loads of space would get saved.

If it helps, I actually use Closure for minification when I minify the modules for publication, and that'll be doing a much better job.

It seems like `par_gh` itself is accessed, just not `0`. Basically a 3 element sparse array is more efficient than defining a 3 element typed array which is why it's done. The element 0 is undefined so shouldn't actually be using much space - however yeah, it's a hangover from the BME680 code it's based on. Shifting everything down by 1 would definitely be an improvement.

AkosLukacs, 2019-01-10:
Yes, it felt like something is not right, so it's better to ask you...

ESLint found an unused `var d = new Uint8Array(len+1);` in the connectSPI function. I think that should be deleted too...
And found some issues in the BME280.js (just took a look at the closest file) as well: for example looks like [setPower](https://github.com/espruino/EspruinoDocs/blob/master/devices/BME280.js#L39) doesn't actually do anything, and it's not mentioned in the docs either.

Ah, of course this is how "oh, I will just take a quick look" :)

Back to the BME680: The minified size numbers are from the Espruino IDE, where can I find the script for the Closure compiler? The docs says an online version is used, not sure if that is up to date.

gfwilliams, 2019-01-11:
This is what's used for minifying: https://github.com/espruino/EspruinoDocs/blob/master/bin/minify.js

But you should be able to use closure as a minify option in the IDE as well...

AkosLukacs, 2019-01-14:
Did some experimenting, for some reason Closure doesn't like the `C` in BME680. With advanced optimizations size comes down to <4k, but the constants are still there (and indeed the `C` object is gone in the SSD1306, maybe there is an object size limit where Closure gives up constant folding?)

Here is a snippet from the optimized code: 

`var a={dc:10,Kb:118,Lb:119,U:97,...` 

And the original code annoted:
```javascript
var C = {
  /** BME680 General config */
  POLL_PERIOD_MS   : 10,       // becomes 'dc', not used in the code

  /** BME680 I2C addresses */
  I2C_ADDR_PRIMARY   : 0x76,   // becomes 'Kb', not used in the code
  I2C_ADDR_SECONDARY : 0x77,   // becomes 'Lb', not used in the code

  /** BME680 unique chip identifier */
  CHIP_ID  : 0x61,             // becomes 'U', this is used in the code
...

```
Don't have the Pixl.js with me, but maybe <4k is good enough. Will try to figure it out tho....

What do you think about about commenting out the unused values in `C`, if can't find a way to get rid of the unused constants at minification time? Should give some extra bytes...

gfwilliams, 2019-01-14:
Thanks for checking - I had a play too, and the weird thing is if you rename it - `C` to `Cx` - so it's not used by anything and then you minify, it gets removed, so I'm not sure it's just size that does it.

But yeah, definitely commenting out seems good - also it looks like creating new private objects works well too and in some cases it makes a lot of sense - like moving `*_MSK` to `MSK.*` allows the whole `MSK` object to be inlined and removed. I just committed some changes along those lines that knock 1.5k off the 'simple' minified size. I imagine there are more things that could be done with - and I guess at some point we'll figure out what's causing `C` not to be collapsed down.

I'm a little iffy about advanced minification because it has caused things to break in the past though...

AkosLukacs, 2019-01-21:
Hi!

Found another quirk: Calling `new BME680()` does issue a [soft reset](https://github.com/AkosLukacs/EspruinoDocs/blob/1611171f4e37a144736694809e88ebf7daaacf20/devices/BME680.js#L214) to the device, and does not complete it's own initialization until after 10ms by reading registers from the physical device.

So if you try to use it immediately, it crashes
```javascript
var bme = bmeLib.connectI2C(i2c);
bme.get_sensor_data());
```
[image]

You have to wait before issuing a reg_sensor_data command.
Added a paragraph to the docs. Hope it's clear...

 - Commenting out the unused constants saves 30 vars.
 - Converted the sparse arrays to regular ones by removing the first `undefined` elements, and changing the indexes, but you were right earlier: that didn't save any memory :(
 - Tried to move some of the constants to new objects like you did with `MSK`, but no luck. There must be some reason, why it's not inlined, but coulldn't figure out...

AkosLukacs, 2019-02-07:
Yes, a callback would be the cleanest solution, but maybe wait for the second time someone runs into this? Or if someone has enough free time :)

Right now the module fits in the Pixl.JS without writing to flash at first, and the code has been running for some time without any problems 👍

closed 2019-02-07 by AkosLukacs