espruino/EspruinoDocs #475: Improving modules in general?

AkosLukacs, 2019-01-10:
Hi!

To avoid mixing all kinds of questions in the BME680 module:
Don't want to push anything on you, but what do you think about using something like ESLint to catch bugs like accidentally creating or modifying globals, and maybe some other issues?

Looks like it's possible to set up ESLint and use it to find some bugs: Found a couple of globals, and other small issues that look like safe to "fix", but wanted to ask you about your opinion before going too deep...

AkosLukacs, 2019-01-11:
Oh, other than global hunting, found couple interesting bugs / issues. Interesting in a sense, that obviously these modules do work.

For example in [NRF24L01P](https://github.com/espruino/EspruinoDocs/blob/master/devices/NRF24L01P.js#L199) `if (!rate in rates) console.log("Unknown rate");` never prints out "Unknown rate", because `!rate` is evaluated first, and becomes `false`. Checked in Espruino, and does the same as in a browser.

[MCP4xxx0](https://github.com/espruino/EspruinoDocs/blob/master/devices/MCP4xxx0.js#L35) the shutdown function writes `value`, but there is no `value` defined.

[UG96](https://github.com/espruino/EspruinoDocs/blob/master/devices/UG96.js#L1135) `resetOptions` becomes or potentially overwrites a global, but is never used.

[aws-iot-device-sdk-device](https://github.com/espruino/EspruinoDocs/blob/master/devices/aws-iot-device-sdk-device.js#L27) `(typeof(value) === null)` is never true.

There are some puzzling ones:
[TouchRD](https://github.com/espruino/EspruinoDocs/blob/master/devices/TouchRD.js#L68) `_p = TRD.prototype` there is no `TRD` anywhere in the whole docs repo?

[MCP4xxx](https://github.com/espruino/EspruinoDocs/blob/master/devices/MCP4xxx.js#L72) `val` and `value` at line 72, 80, 89 how does this work? Also, the [docs] seems to be out-of-date: `setVal` vs `setValue`; `getVal` `getValue`.

In [Pixy camera](https://github.com/espruino/EspruinoDocs/blob/master/devices/Pixy.js) calls to `outBuf` and `outIndex`, `outLen` should be `this.outXX`?

Current progress is [here](https://github.com/AkosLukacs/EspruinoDocs/commit/391ebb198f1cb9e8292e91f0db9e42bff8c3567a), if you are interested. Sorry for creating work for you :)

AkosLukacs, 2019-01-11:
Tried to run the build scripts, but I'm on windows, and don't have node in git-bash, so just started hacking around :)
But have an Ubuntu vm, so should be able to integrate it in the build process.  And wanted to get your feedback as soon as possible. 

The ESLint rules and globals are in the `.eslintrc.json`, disabled and changed to warning some rules, but that's just my "gut-feeling". If you check it, there are still 100+ errors and warnings reported by ESLint, so if you have some time, please take a look at what rules should be disabled, what should be warning, etc.

The easiest way to hack around `npm install -g eslint` (of course, should be installed to the package.json, but...) go to `EspruinoDocs` folder and call `eslint devices/*.js` or `eslint modules/*.js`. If you have VS Code, install the ESLint plugin, and gives immediate feedback as you open a js file, and has intellisense for .eslintrc.json. But there are plugins for other editors as well.

[image]

AkosLukacs, 2019-01-14:
@gfwilliams Your build script is working, the online closure compiler did gave a `Too many compiles performed recently.  Try again later.` error. Did this happen to you?

AkosLukacs, 2019-01-19:
@gfwilliams yes, the build script tried to compile all modules, but got it after some wait...

To run the linter, just `npm install` and `npm run eslint` in the EspruinoDocs folder. Did not add to the build script, because there are still a lot of errors (either errors, or linter rules to be disabled). The devices folder is almost ok. A couple of questions below, but I think that should be merged, if you don't find any more errors.
Not sure if I can fix the rest of the code in modules. Don't want to mess up something can't test...

How do you feel about things like empty blocks? In theory, may help discover unfinished code fragments, but so far only found empty catch blocks. Just disable the rule?

Some file specific questions:

At [Pixy send](https://github.com/espruino/EspruinoDocs/blob/master/devices/Pixy.js#L57) `len` is not defined anywhere. but looks like you at some point you tried to do something with the return value

At [CD4021BE](https://github.com/espruino/EspruinoDocs/blob/master/devices/CD4021BE.js#L60) I guess that should be CD4021BE

At [ESP8266WiFi_0v25](https://github.com/espruino/EspruinoDocs/blob/master/devices/ESP8266WiFi_0v25.js#L328) in `setHostname`: `turnOn` and `MODE` is not defined anywhere?

At [wiolte](https://github.com/espruino/EspruinoDocs/blob/master/devices/wiolte.js#L1004) there are fragments of escaped single quote `"\'"`. Is this necessary for Espruino?

gfwilliams, 2019-01-21:
Thanks! 

Yes, I wouldn't worry about empty blocks at the moment...

Pixy: Yeah, I'd just remove the return value.

CD4021BE: Looks like it! That wasn't tested then :/

ESP8266WiFi_0v25: good catch - that was a copy/paste from EspruinoWiFi. I'd just delete that whole call & error check.

wiolte: Not at all! That was done by Seeed and I don't think they're that used to JS :)

gfwilliams, 2019-01-21:
Just looked through - seems great! Are you ok to do the tweaks above (and re-add the BMP/E280 setPower with the extra line I mentioned)? And then we can keep all these changes together with the right attribution :)

AkosLukacs, 2019-01-21:
Almost there: just noticed that `Pixy.prototype.send` is not used at all. Only `this.spi.send` is actually called. Just remove it?

gfwilliams, 2019-01-21:
I saw that - I wonder if it was really tested, but it seems to be based on http://cmucam.org/projects/cmucam5/wiki/Porting_Guide#The-serial-protocol

I think probably leave it in for now, just in case someone was using it? It's not big and otherwise we're pulling quite a lot of code out!

AkosLukacs, 2019-01-21:
Ok, the tweaks are ready, `npm run eslintdevices` now runs without any issues!

gfwilliams, 2019-02-07:
Closing this now given how much stuff has been fixed!

closed 2019-02-07 by gfwilliams