espruino/EspruinoDocs #480: Missing Publish flags in MQTT library

jjok, 2019-01-16:
Hi. I'd like to publish an MQTT packet that has the `retain` flag set, but it looks like only `qos` is currently implemented.

It would be great to add the retain and DUP flags. I'll see if I can do it myself, but can't promise anything.

http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718038

gfwilliams, 2019-01-16:
Ok, just added. Can you check it out and see if it works?

```
... = require("https://raw.githubusercontent.com/espruino/EspruinoDocs/master/modules/MQTT.js");
mqtt.publish(topic, message, {retain:true});
```

jjok, 2019-01-16:
Thanks! It certainly looks like it should. I'll give it a go tonight, hopefully.

This would be a breaking change for anyone currently passing `qos` as the third param to `mqtt.publish()`. Is that what you're intending or should it check to see if the new `opts` param is just a number, and if it is, assume it's mean to be the `qos`?

gfwilliams, 2019-01-16:
Thanks - I generally try to avoid breaking changes, but in this case:

* it should at least error if it was used
* it wasn't previously mentioned in the docs
* it doesn't appear to be used in any code examples
* the module is based on the API from https://www.npmjs.com/package/mqtt and that doesn't allow it
* everyone always complains about how big the MQTT module is, so anything we can do to make it smaller/more efficient is a bonus :)

jjok, 2019-01-16:
OK. Well, it's fine by me : )

I'm not sure there's a lot you can do to make that module smaller, is there? It looks like it implements the full spec. I can't see there's much you could do to make it smaller, unless you removed features or made micro-optimisations to the way the code is written.

jjok, 2019-01-16:
Hmm... I'm getting an error message. I don't seem to be able to copy it out of the IDE though...

```
Uncaught Error: Cannot read property 'DEF_QOS' of undefined at line 3 col 43

qos: opts.qos || this.C.DEF_QOS
                       ^
in function called from line 14 col 10

in function "subscribe" called from line 13 col 43.
```

jjok, 2019-01-16:
Presumably it isn't `this` when it's inside the callback.

```
topics.forEach(function (topic) {
    subs.push({
        topic: topic,
        qos  : opts.qos || this.C.DEF_QOS
    });
});
```
I can't see another example of where you've had to do that, but I think you should be able to do:

```
var self = this;
topics.forEach(function (topic) {
    subs.push({
        topic: topic,
        qos  : opts.qos || self.C.DEF_QOS
    });
});
```
or
```
topics.forEach(function (topic) {
    subs.push({
        topic: topic,
        qos  : opts.qos || self.C.DEF_QOS
    });
}.bind(this));
```

jjok, 2019-01-16:
Yeah, both of those options work. I'll do a PR.

MaBecker, 2019-01-17:
Just as hint, there is  tinyMQTT module by Olli Phillips

https://github.com/olliephillips/tinyMQTT

gfwilliams, 2019-01-17:
@jjok perfect - thanks for that PR! I think a lot of modules to use either the local var, or bind - there's not one way we use in particular.

@olliephillips are you interested in getting that TinyMQTT module on the Espruino site? I'm happy to add it, and IMO it'd make a lot of sense.

olliephillips, 2019-01-17:
@gfwilliams If you think it would add value, absolutely. 

@jjok pretty sure retain is the default yes

gfwilliams, 2019-01-17:
@olliephillips great - thanks! Just added it and I'll update the website soon at http://espruino.com/tinyMQTT

A bit of a thread hijack, but was there a reason that you didn't define `_q` as a module-local variable rather than global? I just did it here and it seems to get the module size down to 1.38k

jjok, 2019-01-17:
Also not related to this thread...

> Hmm... I'm getting an error message. I don't seem to be able to copy it out of the IDE though...

@gfwilliams Should you be able to copy text out of the left pane of the IDE? It would be pretty useful for copying out error messages.

gfwilliams, 2019-01-17:
I never came across the variable before as a way of saving space, but it's really neat. The change to module scope was just:

```
var _q; // <--- this
function () {
  ...
  _q = this;
}
```

but you wrote the module a while back so maybe it didn't work in Espruino then? Not sure :)

Just a FYI - I was frustrated that the whole `this` token was a waste of space in modules too, so if you turn on `pretokenise` then Espruino automatically converts it and stores it as a single byte - but that doesn't actually help you if you're using 'save on send' with 'to flash' since the text just goes in direct. I'd considered adding something to the IDE to do it before sending but haven't had time yet.

> Should you be able to copy text out of the left pane of the IDE? 

@jjok yes - just click and drag. When you stop the drag then it copies. It's a bit odd but it was done that way because Ctrl-C is used to break out of running code/clear the line.

jjok, 2019-01-17:
> When you stop the drag then it copies

Ahh. Thanks. : )

olliephillips, 2019-01-17:
@gfwilliams I'll look at that and likely change. What minifier do you use, I was using the closure compiler. Also with the docs on Espruino for tinyMQTT, will you pick up changes in say the repo README and reflect on espruino.com, or should I be mindful of this, and update the docs in his repo should the API ever change?

gfwilliams, 2019-01-17:
I was just using closure with simple minification. I wrap in `(function(){....})()` before sending it over - see https://github.com/espruino/EspruinoDocs/blob/master/bin/minify.js.

I'm afraid there's nothing fancy going on with the new page - so if there are any new changes they'll need to be manually mirrored in https://github.com/espruino/EspruinoDocs/blob/master/modules/tinyMQTT.md

It's an interesting thought - I think if I got to the stage where there were a lot of Espruino modules being brought in from their own repos I'd come up with some kind of script, but as far as I remember yours is the only one at the moment.

olliephillips, 2019-01-17:
> I never came across the variable before as a way of saving space, but it's really neat. The change to module scope was just:

Just tested. Why that and adding the closure helps minification is certainly counter-intuitive to me, but it definitely works! Almost 20 more jsvars freed for application code. I'll revise the module to reflect.

MaBecker, 2019-01-17:
> but you wrote the module a while back so maybe it didn't work in Espruino then? Not sure :)

Well Gordon it works perfect from the beginning;-). Just did some small changes to allow receiving 256byte and my latest version can even run from storage after quoting at to sections: [first](https://github.com/MaBecker/tinyMQTT/blob/master/tinyMQTT.js#L68) and [second](https://github.com/MaBecker/tinyMQTT/blob/master/tinyMQTT.js#L102).

Hi @olliephillips, I got some mqtt scenarios and can offer to run them for a couple of days, let me know if you like me to jump in.