espruino/EspruinoDocs #518: NRF52 SAR ADC multi channel measurement not working

AkosLukacs, 2019-08-04:
Hi!

Tried the [Read a buffer of data from the ADC, alternating between 2 pins](https://www.espruino.com/NRF52LL#read-a-buffer-of-data-from-the-adc-alternating-between-2-pins) example, but doesn't seem to sample multiple pins.

Setup: 
 - MDBT42 module, 2v03, running at 2.8V (two AA) 
 - D30 at about ~1.64V
 - D31 at ~1.15V

Running the sample

```javascript
channels : [ { // channel 0
    pin:D30,
    gain:1/4,
    tacq:40,
    refvdd:true,
  }, { // channel 1
    pin:D31,
    gain:1/4,
    tacq:40,
    refvdd:true,
  } ]
```
==> all value in the `buf` around 6800.

Swapping the channels:
``` javascript
channels : [ { // channel 0
    pin:D30,
    gain:1/4,
    tacq:40,
    refvdd:true,
  }, { // channel 1
    pin:D31,
    gain:1/4,
    tacq:40,
    refvdd:true,
  } ]
```
All values are around 9700.

Doing some math, the values by the ADC match the real values:
6800 / 2^14 * 2.8 / 4 * 4 = 1.16V
9700 / 2^14 * 2.8 / 4 * 4 = 1.66V
2.

- 8 / 4 - because Vref is VDD / 4
- * 4 - because gain = 4

So looks like the measurement works, but multiple channels are just ignores.

gfwilliams, 2019-08-05:
I just took another look at this and I noticed that the channel offsets were wrong (12 instead of 16) so I've just fixed that. Still not getting what I expect running the code though :(

gfwilliams, 2019-08-05:
Channel configs look fine too:

```
CH1 0x1500, pin 7, npin 0
CH2 0x1500, pin 8, npin 0
```

So I'm finding it hard to see what's wrong. Worth adding that it seems oversampling doesn't work with >1 channel but the example doesn't do that.

AkosLukacs, 2019-08-05:
Right now I don't want to use scanning, just wanted to try the code sample.

There is this thread in nordic devzone: [SAADC is not scanning](https://devzone.nordicsemi.com/f/nordic-q-a/19133/saadc-is-not-scanning), the answer is _"Local SAADC SAMPLERATE timer can not be used in SCAN mode."_ Honestly, no idea what that means :)
And a link to an [example that does ADC scanning](https://github.com/NordicPlayground/nRF52-ADC-examples/blob/master/ble_app_uart__saadc_timer_driven__scan_mode/main.c) .

gfwilliams, 2019-08-05:
Ahh, ok - thanks! That looks like it then.

I just looked at the datasheet again and while the `SAMPLERATE` register doesn't mention anything, buried in `37.5.2 Continuous mode` it says:

> The SAMPLERATE timer mode cannot be combined with SCAN mode, and only one channel can be enabled in this mode.

So you have to use an external timer with it :/

Just updated the docs and added more checks

commit 4bf2e068 (2019-08-05): Fix #518
closed 2019-08-05 by gfwilliams