espruino/EspruinoDocs #521: boards/Thingy52.md - Pinouts

binloan, 2019-09-13:
Hi,

I'm having a hard time figuring out which actual pins D2/D3 on the Thingy:52 are.
The pins D2/3 are nowhere mentioned in any datasheet.

Thingy | USB->TTL converter
-- | --
GND | GND
D2 | RX ( -> PC )
D3 | TX ( <- PC )
3V | 3.3v (Optional - to run without a battery)

Do you have any P0.XX naming for this? It would make it much easier. Thanks.

gfwilliams, 2019-09-13:
Hi - just replace `D` with `P0.` and you're sorted. I'll update the docs.

gfwilliams, 2019-09-13:
Just to add that this is actually mentioned further up the page...

> Pins P0.02/P0.03/P0.04 on the Thingy PCB correspond to D2/D3/D4

But I'll update the table anyway

closed 2019-09-13 by gfwilliams

binloan, 2019-09-16:
Hi I must have missed that. Thanks for sorting it out!

commit e09f739c (2019-09-13): fix #521