espruino/EspruinoDocs #526: Is there a reason 'nodejs' and note 'node' is used in the build instructions?

mcdonnelldean, 2019-11-21:
This tripped me up trying to build the docs. I was going to submit a PR to change all references of `nodejs` to `node` which is the default. Before I do, is there a reason why it is no-standard?

gfwilliams, 2019-11-22:
Yes - it's because of this: https://launchpad.net/ubuntu/trusty/+package/nodejs-legacy

On some patforms (at least some linux, inc Raspberry Pi) `nodejs` is the default and you require a `nodejs-legacy` package (if it exists) to get `node`.

I guess really `nodejs` should be used everywhere if that's the one that is supported on all platforms - either way consistency would be good

mcdonnelldean, 2019-11-22:
Ok that makes sense! My only concern is if you are a JS dev and coming at this for the first time (eg you want to write a tutorial) you will get tripped up as most folks will have `node` not `nodejs` in their env.

However your point makes sense so it seems like an easy quick win here would be a change to the docs to export nodejs if you get an error. I will make a PR to the README for clarity and see what you think

gfwilliams, 2019-11-22:
Is `nodejs` actually not available on Mac/Windows by default? If so that probably outweighs the Raspberry Pi/Linux and we should just have a note to install `nodejs-legacy` to fix it.

Maybe someone with more knowledge can jump in here but maybe it's not an issue on newer distros - it's fine on Linux Mint 19.1 (based on Ubuntu 18.04) doesn't have a problem.

Personally I prefer `node` anyway.

mcdonnelldean, 2019-11-22:
@gfwilliams Yes indeed `node` is the default on Windows and Mac.

I've asked the question of a few friends who are using Linux to see what there thoughts are.

Happy to make a PR to change it to `node` but it would be nice if it self shimmed, let me come back to this later this evening to see if I can get a nice cross compat shim in so it becomes moot.

I suspect falling down here at first install has likely put a lot of wouldbe contributors off so i'll PR something this evening, even if it's just a notice.

mcdonnelldean, 2019-11-22:
@gfwilliams Thinking about this a little more, given it's a mess to change alias' for the average Windows user but not for a Linux user, I think anote as you mentioned may be the way to go, let me craft a PR and see what your thoughts are

gfwilliams, 2019-11-22:
Thanks - sounds good to me.

The hope is that you can contribute to the documentation without having to get the whole toolchain working though, using just the GitHub Markdown preview.

If there are small issues I fix them after merge though - getting contributions is far more important than having them 100% formatted correctly :)

mcdonnelldean, 2019-11-22:
@gfwilliams We are agreed on that front!

mcdonnelldean, 2019-11-22:
Closed via #528

closed 2019-11-22 by mcdonnelldean