espruino/EspruinoDocs #527: Is it expected that there is minimal CSS when editing offline?

mcdonnelldean, 2019-11-21:
I have successfully built and ran the docs offline, however only a minimal css file has been loaded (offline.css) is this expected or should I expect the full CSS to be loaded on successful build?

gfwilliams, 2019-11-22:
No, that's to be expected, to make it easier to view the pages without any dependencies. You can do `./build.sh --website`, but this will build files assuming they're going to be served by the Espruino website, which adds its own header/footer including CSS.

If you want to make changes to improve offline.css then that'd be great though.

mcdonnelldean, 2019-11-22:
Perfect yes I will make two changes later via PR,

- A little more light styling in offline for reading purposes
- Call out this fact above so that folks know to expect it

Otherwise it all works as expected, I like the markdown additions too, very clever.

mcdonnelldean, 2019-11-22:
@gfwilliams for reference, where is the wider site CSS?

gfwilliams, 2019-11-22:
I don't believe it is in GitHub anywhere.

Main files are:

http://www.espruino.com/css/sitewide.css
http://www.espruino.com/css/prettify.css

And a few hacks in:

http://www.espruino.com/js/sitewide.js

mcdonnelldean, 2019-11-28:
@gfwilliams Great! I just want to bring in a small bit of styling from those to the offline.css to make the offline experience a little less jarring! Should have something for review over the weekend!

mcdonnelldean, 2019-11-28:
Actually @gfwilliams Now that I think about this. My assumption is `offline.css` is not used when the build for the main site is generated. Is that assumption correct?

gfwilliams, 2019-11-28:
Some tweaks to it would be great - and yes, when building for the website it's not included :)

gfwilliams, 2020-11-05:
Closing due to inactivity

closed 2020-11-05 by gfwilliams