espruino/EspruinoDocs #537: boards/MicroBit.md

aidancuckoo, 2020-02-06:
The documentation for the `acceleration()` and `compass()`commands isn't clear - could you please add a program on how to use them?

gfwilliams, 2020-02-06:
Hi - when working correctly the functions will provide an object containing `{x,y,z}` values which you can then use in your code.

```
setInterval(function() {
  var a = acceleration();
  if (a.z>0) show(1)
  else show(2);
}, 100);
```

However, it looks like support hasn't been added for the updated micro:bit's design yet.

Sadly as there's been surprisingly little interest in our work on it (by users or the micro:bit foundation) getting it fixed is quite low priority at the moment.

gfwilliams, 2020-09-21:
This is all fixed now - new micro:bit 1.5 is supported

closed 2020-09-21 by gfwilliams